h2zero/NimBLE-Arduino #643: ESP_ERROR_CHECK failed on platformio, platform=espressif32@6.5.0

Darmandran, 2024-03-04:
sudo mongod --dbpath=/Users/darma/data/db

the project have been confirmed,
will you be able to prepare 

AK-P-46 - 50units
din rail clips - 100units

and 1 samples of
AK-S-99
AK-R-196
AK-R-109
AK-NW-102B
AK-NW-102A
AK-NW-53B
AK-NW-53A
AK-R-177B
AK-N-28
AK-DR-102
AK-R-136
AK-P-45

	AK-S-97
	AK-SP-10
	AK-S-25
	AK-S-79
	AK-R-168

Hi, since beginning of the year i been having issues with the platformio,

```
[env:esp32dev]
platform = espressif32@6.5.0
board = esp32dev 
framework = arduino
board_build.filesystem = littlefs 
monitor_speed = 115200
```

```
08:54:35.508 > ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40094b70
08:54:35.514 > file: ".pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEDevice.cpp" line 879
08:54:35.519 > func: static void NimBLEDevice::init(const string&)
08:54:35.525 > expression: esp_bt_controller_init(&bt_cfg)
08:54:35.527 > 
08:54:35.527 > abort() was called at PC 0x40094b73 on core 1
08:54:35.533 > 
08:54:35.533 > 
08:54:35.533 > Backtrace: 0x40084105:0x3ffced50 0x40094b7d:0x3ffced70 0x4009a8e1:0x3ffced90 0x40094b73:0x3ffcee10 0x400f2236:0x3ffcee30 0x400d3775:0x3ffcee80 0x400d602b:0x3ffceec0 0x40105fda:0x3ffcef20
08:54:35.550 >
```

for platform = espressif32@6.4.0 has no issues

h2zero, 2024-06-04:
No idea what the problem with this would be but 6.7.0 has no such issue. Have you resolved this?

Darmandran, 2024-06-09:
for now i have been sticking with platform = espressif32@6.4.0, and its been good.

h2zero, 2024-06-10:
Thanks, I'll close this then as it seems to be a limited issue.

closed 2024-06-10 by h2zero