h2zero/NimBLE-Arduino #645: no advertising with ESP32-C6

lyusupov, 2024-03-08:
This is a follow up for https://github.com/h2zero/NimBLE-Arduino/issues/642 ticket.

**Branch**: https://github.com/h2zero/NimBLE-Arduino/tree/esp32-c6-test
most recent **Arduino Core** [3.0.0-alpha3](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3)
**NimBLE Server** sketch from examples folder

The boot sequence is fine:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xc24
load:0x4086e610,len:0x2708
load:0x40875728,len:0x594
entry 0x4086c410
Starting NimBLE Server
controller lib commit: [77d09ce]
Advertising Started
```

However, **the ESP32-C6 device is not seen in Nordic tool**.

<br>
<br>

When I build and run the sketch with the same Core but for ESP32-**C3** target - everything looks just fine:

[image]

[image]

[image]

h2zero, 2024-03-09:
Confirmed, error code 0x0207, BLE_ERR_MEM_CAPACITY when setting the advertisement data. 

Will advertise when commenting out the data setting and allow connection but crashes immediately. Looks to be caused by the memory system either not being initialized or missing a step somewhere. I haven't been able to get the debugging to work on the C6 as openocd can't find the config file.

My suspicion is a memory conflict with the controller which is frustratingly closed source so I cannot walk through it.

lyusupov, 2024-03-09:
https://github.com/espressif/openocd-esp32/blob/master/tcl/board/esp32c6-builtin.cfg ?

xelprep, 2024-03-31:
I am having this same issue as well with the esp32-c6-test branch. I'm using this library with a SparkFun ESP32-C6 Qwiic Pocket Development Board which uses a ESP32-C6 Mini-1 SoC. I'm not much of a C dev and Bluetooth is magic to me but please let me know if there's anything I can do to help here. Happy to test or validate any fixes for this library in Arduino or Platformio if needed.

h2zero, 2024-04-06:
This is a pretty major mod to the upstream code and will take time to sort out, I will post here when there are changes pushed for testing.

Jason2866, 2024-05-20:
@btsimonh Not fully correct. Tasmota does use esp-nimble-cpp

h2zero, 2024-06-09:
Pinned this issue to describe the difficulty in supporting future espressif mcu's. They have internally begun using NimBLE for the controller which causes many conflicts that I will try to describe.

- The first important conflict is that since the Arduino core uses bluedroid there necessitates adding NimBLE code to the BT pre-compiled lib for the controller to interface with a bluedroid  host 😢. What happens is the many functions in this repo/NimBLE core files redefine the same functions, okay fine, omit them here in that case.. nope, they are also different and incompatible because espressif changed so much about the porting layer. I'll see what I can do here, any help??? lol

- The next issue is that the config for NimBLE memory has been set by the Arduino core now for these chips, removing the ability to change it and possibly limiting it to the point of being unusable.  No promises that I can work around this one...

- The last issue comes down to maintaining this  repo, the upstream changes have had much more of an effect on keeping this working as a library that it will take considerable time to keep it functional. 

With all of these factors considered, unless espressif or someone with more free time than me can help, I don't think the new devices using NimBLE controllers will ever be supported in a meaningful way. 

The only reasonable way to move this forward that I can see is integration into the Arduino core, perhaps as an option, which I may propose. Otherwise, please use [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp) with Arduino as a component instead of this repo if you need support for the esp32c6, esp32c2, esp32h2, and likely all new BLE enable espressif MCU's 😢

mhaberler, 2024-06-09:
sorry to hear that.. you've done a phenomemal job!

Jason2866, 2024-06-09:
@h2zero One of the many reasons you have listed why we (Tasmota) forked Arduino and IDF and build our own Arduino framework and use your great working `esp-nimble-cpp` together with our forked Arduino. Works great!

h2zero, 2024-06-09:
Thanks for the support! I wouldn't say this repo is dead yet though, still works for the older chips 🙂.

Jason2866, 2024-06-09:
Sorry don't want to be negative. I wrote dead for core > 2.x
Perfectly fine and great when using core 2.x! We used for years with Tasmota.
Thx again for your great work done.

Jason2866, 2024-06-09:
aThere is a silver lightning. espressif Arduino guys are working on a way to make it easy possible to configure and compile the Arduino libs for personal needs.
This will not be done or ready in near future. But this feature is a todo point on there roadmap.

h2zero, 2024-06-09:
That would be great if they did that 🙂.

Just to clarify, this repo still works with core 3.x, just not with the newer chips. 

The interesting thing is I did get it to work as a scanner but advertising seems impossible.

Jason2866, 2024-06-09:
@mhaberler If espressif and Platformio doesn't find an agreement so that Arduino core 3.0.0 is not supported with Platformio, I am considering to do a Platformio Arduino 3.0.0 espressif32 core compatible with `esp-nimble-cpp`. This build would be more compatible, not like the Tasmota framework which has some more breaking changes.

h2zero, 2024-06-12:
I've opened an issue upstream here: https://github.com/espressif/arduino-esp32/issues/9835

Please add your thoughts if you would like.

h2zero, 2024-06-15:
Closing this as C6/H2/C2 and all future BLE Espressif chip support will not be pursued here any further.

closed 2024-06-15 by h2zero