h2zero/NimBLE-Arduino #647: Platformio, arduino and esp32 - pinning nimble to core and debug

SanZamoyski, 2024-03-13:
Hi!

I'm having some problems with my project (it's quite big now, lot of files, so I won't attach it all here, ok?).
I have three files with BLE stuff. In first included to main.cpp, I have:

```
//https://h2zero.github.io/NimBLE-Arduino/md__command_line_config.html (before: 1432941 -> 1432761)
//If defined, NimBLE Client functions will not be included.
#define CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED

//If defined, NimBLE Scan functions will not be included.
#define CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED

//Un-comment to change the core NimBLE host runs on.
#define 	CONFIG_BT_NIMBLE_PINNED_TO_CORE   0

//Un-comment to set the debug log messages level from the NimBLE host stack.
// Values: 0 = DEBUG, 1 = INFO, 2 = WARNING, 3 = ERROR, 4 = CRITICAL, 5+ = NONE
// Uses approx. 32kB of flash memory.
#define 	CONFIG_BT_NIMBLE_LOG_LEVEL   0

#include <NimBLEDevice.h>
```

Then a lot of stuff: advertising, callbacks etc.

Everything seems to work (communication between android apk) but there is no BLE debugging informations.

The problem is I have audioI2S library included and task for audio.loop() pinned to core1. I'm trying to force nimble to pin to core 0. Meybe it's already on core0, as 

```
class MyServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
//      deviceConnected = true;
      Serial.print("[BLE] client connected.");
      lastNotify = millis();
      notifyAll();
      Serial.print(" Running on core ");
      Serial.print(xPortGetCoreID());
    };

    void onDisconnect(NimBLEServer* pServer) {
//      deviceConnected = false;
      Serial.println("[BLE] client disconnected.");
    }
};
```

gives

```
[22:11:39:641] [BLE] client disconnected.␍␊
[22:11:40:307] [BLE] client connected. Running on core 0
```

Is that correct?

SanZamoyski, 2024-03-13:
Ok, got it: had debug=0 in platformio.ini.
How about Core?

h2zero, 2024-03-15:
Yes, the NimBLE host is already on core 0.

If you want to see the NimBLE stack debug logs as well then you will need this: 
```
/** @brief Un-comment to set the debug log messages level from the NimBLE host stack.\n
 *  Values: 0 = DEBUG, 1 = INFO, 2 = WARNING, 3 = ERROR, 4 = CRITICAL, 5+ = NONE\n
 *  Uses approx. 32kB of flash memory.
 */
 // #define CONFIG_BT_NIMBLE_LOG_LEVEL 5
````

closed 2024-03-28 by SanZamoyski