h2zero/NimBLE-Arduino #649: ESP32-S3 crashing with the examples

JanB97, 2024-03-20:
Hello there, unfortunetely I need some help.
I had a already somewhat working programm I made last year. 

I decided to finally pick the programm up again, opened the INO, changed just the BLEDevice name aaaaand my programm crashed the esp32 after flashing. 
I then used a new (never flashed) esp32 and put the server example on it and it crashed again.
I am not really sure whats happening here, the Nimble Version is 1.4.1, also tried older ones.
Arduino IDE is version 2.3.2, I will post the log here if anyone would be so kind a help me figure this out.
I also tried the older BLE implementation, that seems to work (Neil Kolban), but I would like to use nimble to use my existing work.

The crash seems to happen on BLEDevice::startAdvertising();

> 20:00:00.639 -> Rebooting...
> 20:00:00.639 -> �ESP-ROM:esp32s3-20210327
> 20:00:00.686 -> Build:Mar 27 2021
> 20:00:00.686 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> 20:00:00.686 -> Saved PC:0x40377511
> 20:00:00.686 -> SPIWP:0xee
> 20:00:00.686 -> mode:DIO, clock div:1
> 20:00:00.686 -> load:0x3fce3818,len:0x508
> 20:00:00.686 -> load:0x403c9700,len:0x4
> 20:00:00.686 -> load:0x403c9704,len:0xad0
> 20:00:00.686 -> load:0x403cc700,len:0x29d8
> 20:00:00.686 -> entry 0x403c9880
> 20:00:00.771 -> [   138][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42016698
> 20:00:00.805 -> [   150][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42016664
> 20:00:00.805 -> [   161][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42016630
> 20:00:00.805 -> [   172][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420165fc
> 20:00:00.838 -> [   184][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42016698
> 20:00:00.838 -> [   195][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42016664
> 20:00:00.838 -> [   206][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42016630
> 20:00:00.873 -> [   217][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420165fc
> 20:00:00.873 -> [   229][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42016698
> 20:00:00.916 -> [   240][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42016664
> 20:00:00.916 -> [   251][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42016630
> 20:00:00.916 -> [   263][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x420165fc
> 20:00:00.916 -> [   283][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc97914
> 20:00:00.949 -> [   294][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc97914
> 20:00:00.949 -> [   304][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type USB_DM (45) successfully set to 0x42013e10
> 20:00:00.949 -> [   316][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type USB_DP (46) successfully set to 0x42013e10
> 20:00:00.995 -> [   327][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 19 successfully set to type USB_DM (45) with bus 0x3fc9c580
> 20:00:00.995 -> [   338][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 20 successfully set to type USB_DP (46) with bus 0x3fc9c580
> 20:00:03.004 -> =========== Before Setup Start ===========
> 20:00:03.004 -> Chip Info:
> 20:00:03.004 -> ------------------------------------------
> 20:00:03.004 ->   Model             : ESP32-S3
> 20:00:03.004 ->   Package           : 0
> 20:00:03.004 ->   Revision          : 1
> 20:00:03.004 ->   Cores             : 2
> 20:00:03.004 ->   Frequency         : 240 MHz
> 20:00:03.004 ->   Embedded Flash    : No
> 20:00:03.004 ->   Embedded PSRAM    : No
> 20:00:03.004 ->   2.4GHz WiFi       : Yes
> 20:00:03.004 ->   Classic BT        : No
> 20:00:03.037 ->   BT Low Energy     : Yes
> 20:00:03.037 ->   IEEE 802.15.4     : No
> 20:00:03.037 -> ------------------------------------------
> 20:00:03.037 -> INTERNAL Memory Info:
> 20:00:03.037 -> ------------------------------------------
> 20:00:03.037 ->   Total Size        :   374068 B ( 365.3 KB)
> 20:00:03.037 ->   Free Bytes        :   341668 B ( 333.7 KB)
> 20:00:03.037 ->   Allocated Bytes   :    27464 B (  26.8 KB)
> 20:00:03.037 ->   Minimum Free Bytes:   336472 B ( 328.6 KB)
> 20:00:03.037 ->   Largest Free Block:   303092 B ( 296.0 KB)
> 20:00:03.071 -> ------------------------------------------
> 20:00:03.071 -> Flash Info:
> 20:00:03.071 -> ------------------------------------------
> 20:00:03.071 ->   Chip Size         :  8388608 B (8 MB)
> 20:00:03.071 ->   Block Size        :    65536 B (  64.0 KB)
> 20:00:03.071 ->   Sector Size       :     4096 B (   4.0 KB)
> 20:00:03.071 ->   Page Size         :      256 B (   0.2 KB)
[... 77 lines cut ...]
> 20:00:03.478 ->           flags [READ|WRITE]
> 20:00:03.478 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
> 20:00:03.550 -> Debug exception reason: Stack canary watchpoint triggered (ipc0) 
> 20:00:03.550 -> Core  0 register dump:
> 20:00:03.550 -> PC      : 0x4038298e  PS      : 0x00060036  A0      : 0x80380f94  A1      : 0x3fcf0bc0  
> 20:00:03.550 -> A2      : 0x3fc97adc  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
> 20:00:03.550 -> A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80382b21  A9      : 0x3fc992f0  
> 20:00:03.550 -> A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
> 20:00:03.583 -> A14     : 0x03c97a20  A15     : 0x3fc97a20  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
> 20:00:03.583 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
> 20:00:03.583 -> 
> 20:00:03.583 -> 
> 20:00:03.583 -> Backtrace: 0x4038298b:0x3fcf0bc0 0x40380f91:0x3fcf0bf0 0x40382d18:0x3fcf0c20 0x40382d0e:0xa5a5a5a5 |<-CORRUPTED
> 20:00:03.583 -> 
> 20:00:03.583 -> 
> 20:00:03.583 -> 
> 20:00:03.583 -> 
> 20:00:03.583 -> ELF file SHA256: 9f979cf350bee0bb
> 20:00:03.616 -> 
> 20:00:03.616 -> Rebooting...
> 20:00:03.616 -> �ESP-ROM:esp32s3-20210327
> 20:00:03.616 -> Build:Mar 27 2021
> 20:00:03.616 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> 20:00:03.616 -> Saved PC:0x40377511
> 20:00:03.616 -> SPIWP:0xee

JanB97, 2024-03-20:
Okay, I finally figured it out.
I unconsciously updated the ESP32 Board Library to version 3, this seems to break nimble as some apis where changed.

closed 2024-03-20 by JanB97

andrewchilds, 2024-05-08:
Thanks to @JanB97 for posting this. I experienced this as well and downgrading the board to 2.x solved it for me.

@h2zero, just to add more data for this:

Using "esp32" board library version 3.0.0-rc1, the "ESP32S3 Dev Module" board, and the following test code in Arduino IDE version 2.3.2:

```cpp
#include <NimBLEDevice.h>

void setup() {
    Serial.begin(115200);
    delay(1000);
    Serial.println("Begin setup...");

    NimBLEDevice::init("");

    // Delaying here just to allow time to reset/boot device during upload
    delay(10000);

    NimBLEScan* pScan = NimBLEDevice::getScan();
    pScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
    pScan->setActiveScan(true);

    Serial.println("Going to start scanning...");

    pScan->start(5, false); // crash happens here
    Serial.println("esp32 3.0.0-rc will never get here.");
}

void loop() {
}
```

I see the following output:

```
13:58:40.995 -> Begin setup...
13:58:50.356 -> Going to start scanning...
13:58:50.356 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
13:58:50.356 -> Debug exception reason: Stack canary watchpoint triggered (ipc0) 
13:58:50.356 -> Core  0 register dump:
13:58:50.356 -> PC      : 0x40382a9e  PS      : 0x00060036  A0      : 0x80381068  A1      : 0x3fcf0bc0  
13:58:50.356 -> A2      : 0x3fc97c04  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
13:58:50.388 -> A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80382c31  A9      : 0x3fc99420  
13:58:50.388 -> A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
13:58:50.388 -> A14     : 0x03c97b48  A15     : 0x3fc97b48  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
13:58:50.388 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
13:58:50.419 -> 
13:58:50.419 -> 
13:58:50.419 -> Backtrace: 0x40382a9b:0x3fcf0bc0 0x40381065:0x3fcf0bf0 0x40382e28:0x3fcf0c20 0x40382e1e:0xa5a5a5a5 |<-CORRUPTED
13:58:50.419 -> 
13:58:50.419 -> 
13:58:50.419 -> 
13:58:50.419 -> 
13:58:50.419 -> ELF file SHA256: 5f00d74f42bdbacf
13:58:50.419 -> 
13:58:50.419 -> Rebooting...
13:58:50.419 -> ESP-ROM:esp32s3-20210327
13:58:50.419 -> Build:Mar 27 2021
13:58:50.419 -> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
13:58:50.419 -> Saved PC:0x403775a1
13:58:50.451 -> SPIWP:0xee
13:58:50.451 -> mode:DIO, clock div:1
13:58:50.451 -> load:0x3fce3818,len:0x508
13:58:50.451 -> load:0x403c9700,len:0x4
13:58:50.451 -> load:0x403c9704,len:0xad0
13:58:50.451 -> load:0x403cc700,len:0x29d8
13:58:50.451 -> entry 0x403c9880
```

Once I downgraded esp32 from 3.0.0-rc to 2.0.16 the test code worked fine.