h2zero/NimBLE-Arduino #650: Device name not showed at first connection

eccessivo75, 2024-03-21:
Hi, i am experiencing a weird problem. Let me explain.
I have an esp32 based device that, when activated, starts to advertise its name waiting to be connected.
On the other side i have a IOS application that scans for a particular device name and when found it connects the device.
The problem is the first connection on a new (never connected) Iphone.
What happen:
The application try to find the proper device name with no success and so there is no connection
So , i open BLE scanner app on the same phone. After a refresh (or a couple) i am able to see the proper name advertised.
I go back to my app but the device  name is still not found.
Then i go back to BLEscanner, refresh the search and i am able to see the device  name, i push the connect button, and it connects.
After this, i disconnect  blescanner, and go back to my app.... and it connects with no problems, and this will remain true for all the subsequent connection attempt.
I read about a similar issue in the past about the device name , and i think it can be a part of this same issue but i was not able to find a solution. 
I had no this issue with bluedroid.
I there anyone with the same problem ? suggestion?

eccessivo75, 2024-04-01:
It seems that i am the only one having this issue...

h2zero, 2024-04-20:
Sorry, I don't know why this would be other than to say that the device name needs to be advertised in the advertisement data for it to be seen in scanning.

h2zero, 2024-06-09:
Closing this as stale, please reopen if still an issue.

closed 2024-06-09 by h2zero