h2zero/NimBLE-Arduino #652: Changing CPU frequency causes crash

pacmac, 2024-03-23:
As per the title, changing the CPU frequency crashes the device, this is the complete sketch to reproduce.

The device I am using is WROOM32.

```
#include <Arduino.h>
#include <NimBLEDevice.h>
NimBLEScan *pBLEScan;
NimBLEAddress a("cd:d7:31:15:1f:5e");
uint lastAdv = millis();
bool macFound = false;
bool bleScanning = false;

void setup(){
  setCpuFrequencyMhz(20);
  Serial.begin(115200);
  ble_setup();
  bleScan();
}

void loop(){
    Serial.println("Looping Found:" + String(macFound) + " Scanning:" + String(bleScanning)); 
    delay(2000);
    bleScan();
}

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice *advertisedDevice) {
       Serial.println("Found");
      macFound = true;
      lastAdv = millis();
      if(pBLEScan->isScanning()) {
        pBLEScan->stop();
        bleScanning = false;
      }
    }
};

void ble_setup(){
    macFound = false;
    NimBLEDevice::init("");
    NimBLEDevice::whiteListAdd(a);
    Serial.println("Scanning...");
    pBLEScan = NimBLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true);
    pBLEScan->setFilterPolicy(BLE_HCI_SCAN_FILT_USE_WL);
    pBLEScan->setMaxResults(0); // do not store the scan results, use callback only.
}

void bleScan(){
    if(pBLEScan->isScanning() == false) {
      pBLEScan->start(0, nullptr, false);
    }
    bleScanning = true;
}
```

Crash Backtrace:

```
E (42091) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (42091) task_wdt:  - IDLE (CPU 0)
E (42091) task_wdt: Tasks currently running:
E (42091) task_wdt: CPU 0: btController
E (42091) task_wdt: CPU 1: loopTask
E (42091) task_wdt: Aborting.

abort() was called at PC 0x400ea859 on core 0

Backtrace:0x400838f5:0x3ffbe89c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
```

h2zero, 2024-03-24:
Hello, thank you for the detailed issue. Unfortunately this error is related to the btcontroller task which is a closed source binary from espressif and not in the scope of this repo.

pacmac, 2024-03-26:
OK, I see the min freq for WIFI/BT is 80mhz, did not know that.

But while I am here, is it safe to stop scanning the way I have done, does the built in filter have the ability to quit once found ?

h2zero, 2024-04-21:
Yes, that is fine the way you've stopped scanning, there is not an option to do this automatically when found.

pacmac, 2024-04-21:
thanks for confirming that.