h2zero/NimBLE-Arduino #655: how to get faster reconnects from deep sleep

toaster-robotics, 2024-04-04:
Hello,
I currently have a project that uses a ESP32-s3 as a client. It connects with a BLE remote(server) that acts as HID keyboard. I just got everything working and it's catching all the button presses. However, the remote goes into deep sleep after 10 minutes of inactivity. Any button press wakes it up and the ESP32 automatically sees it and reconnects. The issues is that it takes ~2 seconds from starting to completing the reconnection. So any button presses in between those 2 seconds are not caught. This makes it a little awkward to use. Escpeclliy for the non-tech people I am targeting for use. I have a BLE mouse that also goes into deep sleep after a period of inactivity. But it reconnects on my laptop instantly. It seems like sub 1 second. So I know it's possible to reconnect faster. I did a few hours of internet sleuthing and wasn't able to find anything that was helpful. I did come across this [post](https://www.esp32.com/viewtopic.php?t=36025) that references the [blemouse2xac](https://github.com/touchgadget/blemouse2xac) project. However I didn't find anything there that I'm not already doing. Any ideas on how to speed up the reconnections? Thanks

[image]

toaster-robotics, 2024-04-05:
I was able to get the reconnect down to ~0.75 seconds by specifying the **NimBLERemoteCharacteristic** to subscribe to. I have it pared down to 3 **NimBLERemoteCharacteristic** or 3 handles (23, 31, 35). The bulk of the reconnect time is subscribing to the characteristics. One (23) takes ~0.05 seconds to subscribe. But the next two (31, 35) take ~0.35 seconds each. I couldn't get it to go any faster.
One strange thing is that I only need two (23, 31) of the characteristics to catch button presses. But for some reason, if I don't subscribe to the third one (35) I never receive notifications on the other two. If I can get this part figured out then I can cut the time down to ~0.4 seconds.

tapir, 2024-04-05:
Is there a sample code you can share? I'm exactly in the same boat except light sleep instead of deep. Also may I ask how did you manage to wake up from deep sleep with so many keys? AFAIK waking up from deep sleep is only possible with RTC pins which are very limited.

toaster-robotics, 2024-04-06:
@tapir , I just made my code into a [library](https://github.com/toaster-robotics/bleM3Remote). I can't speak for your device. But the remote I'm using wakes with any button press. I have done 0 modifications to the remote. I've only been working on the ESP32 as a client. I don't touch the server. The remote goes for close to $1 on Aliexpress if you want to check it out.

I'm kinda thinking to achieve near instantaneous reconnects from sleep (like on a firestick remote) there would need to be changes to the BLE remote code. But I don't plan on attempting that because: 1. I'm treating it as a off-the-shelf part and 2. I would have no idea how to do that.