h2zero/NimBLE-Arduino #658: notify not updating values in nRF Connect

Gibbz, 2024-04-16:
I'm trying to get my ble notifications to auto update on nRF Connect android app.
Ive selected the button to subscribe to notifications in the app.

I'm unsure if this is a bug with the app, my code, or the library. Basically I have the following snips of code. I assume this is all I need to do to get it to work?

Set characteristic
```
  trip_duration_characteristic = user_data_service->createCharacteristic(
                                         (uint16_t) 0x2BF2,
                                         NIMBLE_PROPERTY::READ |
                                         NIMBLE_PROPERTY::NOTIFY);
```

Apply a value, and notify
```
    trip_duration_characteristic->setValue(value);
    trip_duration_characteristic->notify();
```

h2zero, 2024-04-20:
Your code looks fine, could you set the log level to debug and share logs of the notification being sent?

h2zero, 2024-06-09:
Closing this as there has been no response, please reopen if still and issue.

closed 2024-06-09 by h2zero