h2zero/NimBLE-Arduino #659: Is it possible to advertise the device name as a client?

sepp89117, 2024-04-22:
Hi,
I'm trying to make the device name readable with an ESP32 BLE client with UUID 0x2a00.
The BLE server (GoPro camera) tries to read the device name (0x2a00) after pairing.
The client (ESP32 with NimBLE) sends back an error response 'attribute not found'.
I tried adding the following code in the setup, but the response to the GoPro's request remains 'Attribute not found':
```cpp
NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
pAdvertising->setName(DEVICE_NAME);
pAdvertising->start();
```

Log output:
```
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
D NimBLEAdvertising: << Advertising start
```

Wireshark recording excerpt:
[image]

Is there a way to do this with ESP32 and NimBLE? When pairing with a smartphone, the GoPro can read the device name of the smartphone.

Thanks a lot!

h2zero, 2024-04-22:
You can do this by creating a server instance, do this before creating any clients:
```
NimBLEDevice::init("devicename");
NimBLEServer * pServer = NimBLEDevice::createServer();
pServer->start();
```

sepp89117, 2024-04-23:
Thank you! I actually should have figured that out myself, sorry!
After including "NimBLEServer.h" your code worked.

Thanks again!

closed 2024-04-23 by sepp89117