h2zero/NimBLE-Arduino #661: How to achieve long-distance connection communication？

Blue186, 2024-04-25:
Hello, 
what should I do to extend the Bluetooth connection distance? In terms of software, I set the maximum gain `NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */`
but the distance is only about 8m, so the connection is unstable, but my Bluetooth antenna can support a connection distance of about 100m. (Using the ESP32 BLE basic library); Are there any other parameter settings that can achieve stable long-distance connections?

h2zero, 2024-04-25:
Are you saying that with the bluedroid library you get more distance than with NimBLE? I don't know how that could be but please test and confirm so I can look at it. FWIW I get well over 30m with my esp32's at 9dbm.

Blue186, 2024-04-26:
Thanks for answering my question，
It should be that my other configuration parameters were not done well. I found that by adjusting the parameters of `pClient->updateConnParams();,` I can adjust the connection distance, and finally successfully achieved a 100-meter connection.
[image]
At the same time, in addition to the updateConnParams function on the client, the server also has it. What configurations should I make on the client and server to achieve a stable and efficient connection between the client and the server? The main ones are connection distance and data delay.

h2zero, 2024-04-26:
That makes sense, you should be loose with the timeout for distance, latency also if you're inclined. The min max interval being a factor of 15ms is recommended by Apple but there is no reason to follow that other than connecting with Apple devices that I can fathom. Trial and error here will be required to maximize if that's the goal.

closed 2024-06-08 by h2zero