h2zero/NimBLE-Arduino #665: How to obtain client from existing server connection?

gkoh, 2024-05-04:
### Use Case
NimBLE device is configured as a server and advertises.
Smartphone client scans, can see peripheral device name and connects.

After connection, would like server to use existing connection to query (and display) client device name via service `0x1800` and characteristic `0x2A00`.

### What I've tried
* record peer ota address during server `onConnect()`
   * try `NimBLEClient::connect()` to this address
   * get 'already connected' due to `ble_gap_conn_find_by_addr()` finding the existing underlying connection
 * looked for others way to get a valid client object from the existing server handle, failed

### What I hacked to prove it's possible
* modified `NimBLEClient::connect()` to:
    * use the `ble_gap_conn_desc` output of `ble_gap_conn_find_by_addr()` check
    * make the `NimBLEClient` connected with the details from the gap conn desc
    * used this client to get remote service and characteristic from the central
    * success

### Possible Solutions
1. Nothing, I'm doing something crazy and weird
2. Modify `NimBLEClient::connect()` to allow and succeed with an already connected OTA address
3. Add a method to `NimBLEServer` for getting a properly connected `NimBLEClient` based on the underlying connection
4. Something else?

h2zero, 2024-06-04:
Work in progress here: https://github.com/h2zero/esp-nimble-cpp/pull/156

h2zero, 2024-06-04:
API added to master branch here: https://github.com/h2zero/NimBLE-Arduino/commit/6db036830c58ae85e3f691ceebada96e6d8ba4af

closed 2024-06-04 by h2zero

gkoh, 2024-06-05:
> API added to master branch here: [6db0368](https://github.com/h2zero/NimBLE-Arduino/commit/6db036830c58ae85e3f691ceebada96e6d8ba4af)

Nice!
Once again, much thanks and appreciation.