h2zero/NimBLE-Arduino #668: How to get current PHY mode

henkelis, 2024-05-24:
Hi,

For my use case I want to get the current mode of a connection, to be able to display it (from the client side, but could also be from the server side). I also want to be able to change the mode after connection.

I've added functions to NimbleClient and NimbleServer to do that as follows (based on existing functions):

- getLePhy - calls ble_gap_read_le_phy in ble_gap
- setLePhy - calls ble_gap_set_prefered_le_phy in ble_gap

The functions work and I can get and set modes. Unfortunately when getting the modes for coded PHY, the BLE spec states to return just that it is coded, and not the actual mode (S2 or S8). The mode appears to be available in the low level connection data, but I can't work out how to get to that. 

I can see HCI code that gets the PHY (ble_ll_conn_hci_le_rd_phy) and could cheat and return the mode in the top 4 bits alongside the phy, but my board (Adafruit Feather esp32-s3) provides precompiled bt libs so I would have to edit that source and recompile those libs.

Is there another way to get the actual mode?

Mark.

h2zero, 2024-05-24:
Hello, you can assume it to be S8, S2 is almost never used.

henkelis, 2024-05-24:
All the more reason to be able to detect if it's set :)