h2zero/NimBLE-Arduino #670: Name clash of `const char *TAG` with Arduino core macro `TAG`.

everslick, 2024-05-28:
Easy to fix by renaming **TAG** to e.g. **tag** in:

nimble/esp_port/esp-hci/src/esp_nimble_hci.c
nimble/nimble/host/store/config/src/ble_store_nvs.c

h2zero, 2024-05-30:
I've never heard of this issue before, this must be something to do with the new arduino core?

everslick, 2024-05-31:
Not really, this is a feature that was implemented with PR https://github.com/espressif/arduino-esp32/pull/4845 in 2021.

It allows redirecting logs from IDF and Arduino to your own logging mechanism by calling `esp_log_set_vprintf()` and defining `USE_ESP_IDF_LOG`. Additionally it allows setting the `TAG` define to give logs from IDF a unique tag name which the clashes with the `TAG` const in NimNLE-Arduino.

I also first thought it was CORE 3.0.0 that broke my build and reported this in Arduino CORE:
https://github.com/espressif/arduino-esp32/issues/9704.

everslick, 2024-05-31:
Please note that renaming `TAG` to something else does fix the build when a user defines `USE_ESP_IDF_LOG`, so at least the firmware compiles. But to make the LOG redirection work also for Nimble-Arduino, I had to make a small change in **NimBLELog.h**:

```
#if defined(CONFIG_NIMBLE_CPP_IDF) || defined(CONFIG_NIMBLE_CPP_LOG_IDF) // using esp-idf
#  include "esp_log.h"
#  ifndef CONFIG_NIMBLE_CPP_LOG_LEVEL
#    define CONFIG_NIMBLE_CPP_LOG_LEVEL 0
#  endif

#  define NIMBLE_CPP_LOG_PRINT(level, tag, format, ...) do { \
    if (CONFIG_NIMBLE_CPP_LOG_LEVEL >= level) \
      ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
    } while(0)
```

I introduced the new `CONFIG_NIMBLE_CPP_LOG_IDF` define and I had to replace `ESP_LOG_LEVEL_LOCAL` with `ESP_LOG_LEVEL`. Now I could add:

```
#define CONFIG_NIMBLE_CPP_LOG_IDF
#define CONFIG_NIMBLE_CPP_LOG_LEVEL CORE_DEBUG_LEVEL
```

to **nimconfig.h** and all is good. ;-)

I'm not sure if this is the cleanest or correct way to do it, but now NimBLE logs also get logged correctly via my own logging mechanism.

everslick, 2024-06-04:
Thank you very much! Also for maintaining this NimBLE fork! (which should be the canonical BLE lib in Ardunio, IMHO!)

PR #682 (2024-06-13): Rename "TAG" to "LOG_TAG" to avoid conflict with Arduino core.
Fixes #670
commit 068419d9 (2024-06-13): Rename "TAG" to "LOG_TAG" to avoid conflict with Arduino core.

Fixes #670
closed 2024-06-13 by h2zero