h2zero/NimBLE-Arduino #672: Error compile Arduino BLE 5 example for ESP32

alberk8, 2024-05-30:
Hi,

I am getting the following error using the [sample code](https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/examples/Bluetooth_5/NimBLE_multi_advertiser/NimBLE_multi_advertiser.ino).  Compiling using Arduino IDE 2.3.2 for ESP32 C3.   What is missing here? 

```
d:\Staging\Arduino\libraries\NimBLE-Arduino\src\nimble\nimble\host\src\ble_hs_periodic_sync.c:58:24: error: 'struct ble_npl_event' has no member named 'event'
     if((psync->lost_ev).event != NULL)

```

h2zero, 2024-05-30:
@alberk8, what arduino core version are you using?

alberk8, 2024-05-31:
@h2zero ,  It is using Esp32 2.0.11.   Upgrading to 2.0.17 also did not work.

h2zero, 2024-06-04:
Do  you have periodic advertisements enabled? if so please disable and try again. I haven't had time to implement this yet.

alberk8, 2024-06-04:
Disabling the two lines in the nimconfig.h enable compilation of the code.   Does the sample code actually extend the BLE 5 range?

```
/** @brief Un-comment to enable periodic advertising */
// #define CONFIG_BT_NIMBLE_ENABLE_PERIODIC_ADV 1

/** @brief Un-comment to change the maximum number of periodically synced devices */
// #define CONFIG_BT_NIMBLE_MAX_PERIODIC_SYNCS 1
```

h2zero, 2024-06-04:
Great, yes it does extend it a good amount depending on your device/antenna.