h2zero/NimBLE-Arduino #675: Error in converting EddystoneTLM negative temperatures

giovanni-foorban, 2024-06-06:
In [NimBLEEddystoneTLM.cpp](https://github.com/h2zero/NimBLE-Arduino/blob/nimble-1.5.0/src/NimBLEEddystoneTLM.cpp) there is an issue when receiving negative numbers for temperature readings and when setting temperature.

For example: -0,1C is converted to approximately 256C.

On our side we fixed the issue like this:
```diff
/**
 * @brief Set the temperature to advertise.
 * @param [in] temp The temperature value.
 */
void NimBLEEddystoneTLM::setTemp(float temp) {
-  m_eddystoneData.temp = (uint16_t)temp;
+  m_eddystoneData.temp = ENDIAN_CHANGE_U16((int16_t)(temp * 256.0f)) ;
} // setTemp
/**
 * @brief Get the temperature being advertised.
 * @return The temperature value.
 */
float NimBLEEddystoneTLM::getTemp() {
-    return ENDIAN_CHANGE_U16(m_eddystoneData.temp) / 256.0f;
+ return (int16_t)ENDIAN_CHANGE_U16(m_eddystoneData.temp) / 256.0f;
} // getTemp
```

h2zero, 2024-06-06:
Thanks! would you mind making a PR for this?

Max93, 2024-06-06:
Hello @h2zero 

I am a colleague of Giovanni (@giovanni-foorban), who opened this issue. I have created a pull request to address the problem with negative temperature readings #677.

You can find the pull request here.

Thank you for your attention.

closed 2024-06-08 by h2zero

Max93, 2024-06-09:
Hi @h2zero, when are you planning the next release ? Was it right to open the pr to release 1.4 ?

h2zero, 2024-06-09:
I'll look to make a new release soon, hard to say when exactly