h2zero/NimBLE-Arduino #676: ESP32-S3 crash with SDK 3.x.x

Laxilef, 2024-06-06:
Hi
I'm using NimBLE-Arduino 1.4.1 with SDK 2.x.x and it works great, but now I'm trying to upgrade to SDK 3.x.x and when I call the ``NimBLEClient::connect()`` method I get a crash.
The error always looks like this:
```
Guru Meditation Error: Core 0 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (ipc0)
```

I tried increasing the task stack to 50k, it doesn't help.
Platformio.ini config:
```ini
[esp32_defaults]
framework = arduino
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
```

This is very similar to #649. Does this mean that the library is currently not compatible with SDK 3.x.x?

h2zero, 2024-06-06:
Hello @Laxilef, thanks for opening this issue. Yes, the new arduino core has broken this library for now and I will fix this in the future. It will still work but it requires modifying your application to do the BLE related calls from a task running on the same core as the main NimBLE stack task like so:
```
void BLETask(void* param) {
  for(;;) {
    ..BLE STUFF GOES HERE..
    delay(1); // always delay in the loop to allow other tasks to run
  }
  vTaskDelete(NULL); // should never get here
}

void setup() {
  ...YOUR INIT CODE...
  xTaskCreatePinnedToCore(
     BLETask, /* Function to implement the task */
     "BLETask", /* Name of the task */
     4096, /* Stack size in bytes */
     NULL, /* Task input parameter */
     2, /* Priority of the task (set higher than loop) */
     nullptr, /* Task handle. */
     CONFIG_BT_NIMBLE_PINNED_TO_CORE); /* Core where main nimble task runs */
}

void loop() {
  delay(1);
}
```

Laxilef, 2024-06-06:
Yeah, it works! Thanks @h2zero )

p.s. did you find any other bugs with 3.x.x?

svdrummer, 2024-06-06:
Unrelated, however, ESP-NOW had some breaking changes made. Espressif has noted them, but the Arduino wrapper team missed it.

h2zero, 2024-06-06:
@Laxilef Im not aware of anything else, I just knew the build broke due to IPC changes.

I've pinned this for everyone to comment on their experience and to make the workaround visible.

ednieuw, 2024-06-10:
I will wait for the fix and go back to esp32 Version 2.

h2zero, 2024-06-10:
Please test #681 and let me know if it resolves this for you all.

Hougaard22, 2024-06-10:
> Please test #681 and let me know if it resolves this for you all.

The change resolves the issue, I am now able to scan without getting the `Core 0 panic'ed` error 👍

ednieuw, 2024-06-11:
Same on ESP32-S3WROOM-1 Dev board

ednieuw, 2024-06-11:
Using the last library with the Arduino Nano ESP32 board that still uses esp32 2.0.13 gave these errors during compiling

```

C:\\Users\\ednie\\AppData\\Local\\Temp\\arduino\\sketches\\8800A92742734354C2C4BFDC9F9ADF6B/ESP32Arduino_WordClockV029.ino.elf"
c:/users/ednie/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ednie\AppData\Local\Temp\arduino\sketches\8800A92742734354C2C4BFDC9F9ADF6B\sketch\objs.a(ESP32Arduino_WordClockV029.ino.cpp.o):(.literal._Z15StartBLEServicev+0x20): undefined reference to `NimBLEAdvertising::start(unsigned int, void (*)(NimBLEAdvertising*), NimBLEAddress*)'
c:/users/ednie/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ednie\AppData\Local\Temp\arduino\sketches\8800A92742734354C2C4BFDC9F9ADF6B\sketch\objs.a(ESP32Arduino_WordClockV029.ino.cpp.o): in function `StartBLEService()':
C:\Users\ednie\Documents\Files\Arduino\Arduino_Nano-ESP32-WordClock\ESP32Arduino_WordClockV029/ESP32Arduino_WordClockV029.ino:2591: undefined reference to `NimBLEAdvertising::start(unsigned int, void (*)(NimBLEAdvertising*), NimBLEAddress*)'
collect2.exe: error: ld returned 1 exit status

```

h2zero, 2024-06-11:
@ednieuw Have you tested with the PR mentioned above ? #679

ednieuw, 2024-06-11:
You BLE-uart example also gives the same error as above and  the Arduino Nano ESP32 does not initializes the BLE serial and simply stops without rebooting

ednieuw, 2024-06-11:
BTW The Nano was also rebooting
FYI I used this piece of code (from the Espressif  migration 2.x -> 3.x document)  to make a library compatible for both esp32 versions

```
#ifdef ESP_ARDUINO_VERSION_MAJOR
            #if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
                                                  // Code for version 3.x

                   #else
                                                 // Code for version 2.x
                   #ifdef ESP32

                   #else
 

                    #endif
                 #endif
                #endif
```

h2zero, 2024-06-11:
Sorry I mentioned the wrong PR... the fix should be in #681

ednieuw, 2024-06-11:
Now I have NimBLE-Arduino-release-1.4.zip

I tested that this morning but I will test it again to be sure because there is also NimBLE-Arduino-master.zip in my download folder

ednieuw, 2024-06-11:
No. still rebooting
checked esp_nible_hci.c to be sure I had the right library version 
Saw all the Github changes. 
For example at line 75
```
/* Added; Called from the core NimBLE is running on, not used for unicore */
#ifndef CONFIG_FREERTOS_UNICORE
void ble_hci_trans_hs_cmd_tx_on_core(void *arg)
{
    // Ugly but necessary as the arduino core does not provide enough IPC stack for variables.
    esp_vhci_host_send_packet((uint8_t*)arg, *((uint8_t*)arg + 3) + 1 + BLE_HCI_CMD_HDR_LEN);
}
#endif
```

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037790d
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
Serial started
Mem.Checksum = 25065
Setting loaded
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (ipc0) 
Core  0 register dump:
PC      : 0x403835aa  PS      : 0x00060036  A0      : 0x80381a9c  A1      : 0x3fcf0d00  
A2      : 0x3fc986ac  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x8038373d  A9      : 0x3fc9ae00  
A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x03c985a8  A15     : 0x3fc985a8  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x403835a7:0x3fcf0d00 0x40381a99:0x3fcf0d30 0x40383934:0x3fcf0d60 0x4038392a:0xa5a5a5a5 |<-CORRUPTED
```

ELF file SHA256: 10f61709dc4a6b25

`

h2zero, 2024-06-11:
If you still have this:
```
/* Added; Called from the core NimBLE is running on, not used for unicore */
#ifndef CONFIG_FREERTOS_UNICORE
void ble_hci_trans_hs_cmd_tx_on_core(void *arg)
{
    // Ugly but necessary as the arduino core does not provide enough IPC stack for variables.
    esp_vhci_host_send_packet((uint8_t*)arg, *((uint8_t*)arg + 3) + 1 + BLE_HCI_CMD_HDR_LEN);
}
#endif
```
in esp_nible_hci.c then you do not have the PR branch as that is what gets removed there.

ednieuw, 2024-06-11:
Can you send the library bij mail to me or direct me how to download it?

Laxilef, 2024-06-11:
> Please test #681 and let me know if it resolves this for you all.

Tested on esp32 s3, core 3.0.1, esp-idf v5.1.4, works great

ednieuw, 2024-06-11:
Thanks for the link Laxilef! I still can not find it in Github but with your link I got success
The code works on ESP32 S3 with core 3.0.1 

But now an error with esp32 core 2.0.13 compiling for Arduino Nano ESP32
Would be nice if it work on both versions.

```

c:/users/ednie/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ednie\AppData\Local\Temp\arduino\sketches\8800A92742734354C2C4BFDC9F9ADF6B\sketch\objs.a(ESP32Arduino_WordClockV029.ino.cpp.o):(.literal._Z15StartBLEServicev+0x20): undefined reference to `NimBLEAdvertising::start(unsigned int, void (*)(NimBLEAdvertising*), NimBLEAddress*)'
c:/users/ednie/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\ednie\AppData\Local\Temp\arduino\sketches\8800A92742734354C2C4BFDC9F9ADF6B\sketch\objs.a(ESP32Arduino_WordClockV029.ino.cpp.o): in function `StartBLEService()':
C:\Users\ednie\Documents\Files\Arduino\Arduino_Nano-ESP32-WordClock\ESP32Arduino_WordClockV029/ESP32Arduino_WordClockV029.ino:2587: undefined reference to `NimBLEAdvertising::start(unsigned int, void (*)(NimBLEAdvertising*), NimBLEAddress*)'
collect2.exe: error: ld returned 1 exit status

```

h2zero, 2024-06-12:
@ednieuw have you set `CONFIG_BT_NIMBLE_ROLE_BROADCASTER_DISABLED` to 1 somewhere?

h2zero, 2024-06-13:
@ednieuw I'm not able to reproduce this issue, there must be something in your environment causing this.

h2zero, 2024-06-13:
That sounds likely, please let me know. I recommend switching to PlatformIO in vscode as it makes this sort of thing much easier to manage.

ednieuw, 2024-06-13:
Succes. Both ESP32 version V2.0 and v3.0 work now. I tried the ESP32-S3 with V3.0.1 and the Arduino Nano ESP32 with the Arduino version V2.0 13.
I will give platformIO a new  try. I was not very happy with it a few years ago. 
Thanks for your support

h2zero, 2024-06-13:
Great! thanks for the update! I will close this now.

closed 2024-06-13 by h2zero