h2zero/NimBLE-Arduino #685: The future of this repo... new espressif MCU's cannot be supported.

h2zero, 2024-06-15:
My apologies to all, I cannot find a workaround for the upstream changes that have affected this repo's ability to work with arduino-esp32 core and the new C6/H2/C2 MCU's. The issue I created https://github.com/espressif/arduino-esp32/issues/9835 has been relegated to a discussion, where there is essentially no visibility, so I have now closed it 😞.

This marks the end of my support for NimBLE on the esp32 platform for Arduino, I will continue to maintain this as best as I can for the chips that can use it, the latest Espressif devices and future one will not be supported here, unless I can find a way to get around the roadblocks.

I suggest to you all... plead actually... move on from Arduino, it's great to start from, but you can do better. Use IDF directly and esp-nimble-cpp with Arduino as a component if you want to use the work here, as I will always be able to support this via wrapping the underlying NimBLE stack. Better yet use PlatformIO with idf and arduino and link esp-nimble-cpp as a lib_dep :) I'll make some instruction for this in the near term.

Thank you all for the incredible support, I've put in thousands of hours to make this work and it always puts a big smile on my face to see people recommending/using and submitting issues to this repo.

h2zero, 2024-08-10:
Thank you @afpineda for your support! I'm incredibly grateful to all of you for the support and contributions.

lyusupov, 2024-08-10:
## For anyone who may consider it useful

I've built [**ArduinoBLE**](https://github.com/arduino-libraries/ArduinoBLE/releases/tag/1.3.7) recently for ESP32-C6 target  and **Arduino Core 3.0.4**. It needs to apply a minor patch: https://github.com/arduino-libraries/ArduinoBLE/pull/364

[**My use case**](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/ArduinoBLE.cpp) (HM-10 compatible BLE UART) boots and works nicely on ESP32-C6 afterwards.
The same app can work with 
- [Bluedroid](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/Bluedroid.cpp)  - all the ESP32-XX
- [NimBLE](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/NimBLE.cpp) - ESP32, ESP32-C3 and ESP32-Sx only (Arduino Core 2.x).

Positive effect of using ArduinoBLE as a replacement for Bluedroid with ESP32-C6 and **Arduino Core 3.0.4**  is saving approx. 450 Kbytes of flash memory space and approx. 40 Kbytes of RAM.

troglodisme, 2024-08-20:
> ## For anyone who may consider it useful
> I've built [**ArduinoBLE**](https://github.com/arduino-libraries/ArduinoBLE/releases/tag/1.3.7) recently for ESP32-C6 target and **Arduino Core 3.0.4**. It needs to apply a minor patch: [arduino-libraries/ArduinoBLE#364](https://github.com/arduino-libraries/ArduinoBLE/pull/364)
> 
> [**My use case**](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/ArduinoBLE.cpp) (HM-10 compatible BLE UART) boots and works nicely on ESP32-C6 afterwards. The same app can work with
> 
> * [Bluedroid](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/Bluedroid.cpp)  - all the ESP32-XX with exception of S2
> * [NimBLE](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/bluetooth/NimBLE.cpp) - ESP32, ESP32-C3 and ESP32-S3 only (Arduino Core 2.x).
> 
> Positive effect of using ArduinoBLE as a replacement for Bluedroid with ESP32-C6 and **Arduino Core 3.0.4** is saving approx. 450 Kbytes of flash memory space and approx. 40 Kbytes of RAM.

@lyusupov I am very interested. Did you test NimBLE on ESP32-C6 BLE without using the HM-10? 
I have managed to build ArduinoBLE applying your suggested patch, but not sure how to modify NimBLE to make it work. 

Could you please help me in the process? If you are available, please let me know how I can get in touch. Thanks

lyusupov, 2025-02-05:
https://github.com/espressif/arduino-esp32/pull/10937

lyusupov, 2025-02-20:
https://github.com/espressif/arduino-esp32/discussions/10991

closed 2025-04-23 by h2zero