h2zero/NimBLE-Arduino #686: NimBLE-Arduino crashes with esp32 version 3.0.1. and a LilyGo T-display S3

Berg0162, 2024-06-16:
Board: LilyGo T-display S3 --> Board manager setting: "ESP32S3 Dev Module" with the right Tools settings
Arduino IDE version: 2.3.2; CLI Version: 0.35.3

Arduino-ESP32 version: 2.0.17 and previous
--> All my code compiles without problems and works fine with NimBLE-Arduino (version 1.4.1) !!

After installing: Arduino-ESP32 version 3.0.1 my code crashes when `pAdvertising->start();` is called...

Checked with <examples/NimBLE_Server.ino> and your NimBLE-Server example crashes at `pAdvertising->start();`

```C++
    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    /** Add the services to the advertisment data **/
    pAdvertising->addServiceUUID(pDeadService->getUUID());
    pAdvertising->addServiceUUID(pBaadService->getUUID());
    /** If your device is battery powered you may consider setting scan response
     *  to false as it will extend battery life at the expense of less data sent.
     */
    pAdvertising->setScanResponse(true);
    pAdvertising->start();

    Serial.println("Advertising Started");
```

```
09:25:43.471 -> Saved PC:0x4037770d
09:25:43.471 -> SPIWP:0xee
09:25:43.471 -> mode:DIO, clock div:1
09:25:43.471 -> load:0x3fce3818,len:0x508
09:25:43.471 -> load:0x403c9700,len:0x4
09:25:43.471 -> load:0x403c9704,len:0xad0
09:25:43.471 -> load:0x403cc700,len:0x29e4
09:25:43.471 -> entry 0x403c9880
09:25:43.610 -> Starting NimBLE Server
09:25:43.737 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
09:25:43.737 -> Debug exception reason: Stack canary watchpoint triggered (ipc0) 
09:25:43.737 -> Core  0 register dump:
09:25:43.737 -> PC      : 0x40383952  PS      : 0x00060036  A0      : 0x80381f58  A1      : 0x3fcf0980  
09:25:43.737 -> A2      : 0x3fc98820  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
09:25:43.737 -> A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80383ae5  A9      : 0x3fc9a0d0  
09:25:43.737 -> A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
09:25:43.737 -> A14     : 0x03c98758  A15     : 0x3fc98758  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
09:25:43.737 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
09:25:43.737 -> 
09:25:43.737 -> 
09:25:43.737 -> Backtrace: 0x4038394f:0x3fcf0980 0x40381f55:0x3fcf09b0 0x40383cdc:0x3fcf09e0 0x40383cd2:0xa5a5a5a5 |<-CORRUPTED
```

Checked (esp32 3.0.1) with a different board: Adafruit Feather esp32 V2 --> NO crash --> your NimBLE-Server example works fime !!

Berg0162, 2024-06-16:
I have tested (the ESP32S3 Dev Module) with your `examples/NimBLE_Client.ino` and when `pScan->start(scanTime, scanEndedCB);` is called the code crashes 
```C++
    /** Active scan will gather scan response data from advertisers
     *  but will use more energy from both devices
     */
    pScan->setActiveScan(true);
    /** Start scanning for advertisers for the scan time specified (in seconds) 0 = forever
     *  Optional callback for when scanning stops.
     */
    pScan->start(scanTime, scanEndedCB);
}
```

```
10:43:50.729 -> Rebooting...
10:43:50.729 -> ESP-ROM:esp32s3-20210327
10:43:50.729 -> Build:Mar 27 2021
10:43:50.764 -> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
10:43:50.764 -> Saved PC:0x4037770d
10:43:50.764 -> SPIWP:0xee
10:43:50.764 -> mode:DIO, clock div:1
10:43:50.764 -> load:0x3fce3818,len:0x508
10:43:50.764 -> load:0x403c9700,len:0x4
10:43:50.764 -> load:0x403c9704,len:0xad0
10:43:50.764 -> load:0x403cc700,len:0x29e4
10:43:50.764 -> entry 0x403c9880
10:43:50.903 -> Starting NimBLE Client
10:43:51.019 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
10:43:51.019 -> Debug exception reason: Stack canary watchpoint triggered (ipc0) 
10:43:51.019 -> Core  0 register dump:
10:43:51.019 -> PC      : 0x4038398e  PS      : 0x00060036  A0      : 0x80381f58  A1      : 0x3fcf0980  
10:43:51.019 -> A2      : 0x3fc98818  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
10:43:51.019 -> A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80383b21  A9      : 0x3fc9a0d0  
10:43:51.019 -> A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
10:43:51.019 -> A14     : 0x03c98750  A15     : 0x3fc98750  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
10:43:51.019 -> EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
10:43:51.019 -> 
10:43:51.019 -> 
10:43:51.019 -> Backtrace: 0x4038398b:0x3fcf0980 0x40381f55:0x3fcf09b0 0x40383d18:0x3fcf09e0 0x40383d0e:0xa5a5a5a5 |<-CORRUPTED
```
I would very much appreciate when you have a look into this problem with esp 3.0.1. It would not be the first time that our friends with esspressif spoil the party with changes....

h2zero, 2024-06-16:
Please use the latest commit from the release/1.4 branch as it contains the fix already. I'll make a new release shortly.

Berg0162, 2024-06-17:
Thank you so much! I will keep esp32 to version 2.0.17 until the new NimBLE-Arduino version is released. Release esp32 3.0.0 turned out to crash the TFT_eSPI library as well... One surprise after the other!

h2zero, 2024-06-23:
Closing this as 1.4.2 has been released with the fix.

closed 2024-06-23 by h2zero