h2zero/NimBLE-Arduino #687: NimBLE-Arduino debug exception when trying to initilialize BLE functions

capc0307, 2024-06-17:
Hi,

I'm having trouble with the library. I have used the exact logic of my code on a project not to long ago and it worked well. Now I'm implementing the exact logic in a considerable larger project and it's thowing the following debug exception. What does this debug excepcition mean and how can I solve it? I'm currently using 1.4.1 version of the library and a ESP32-S3 with the version 3.0.1 of the _Arduino core for ESP32_.

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (ipc0) 
Core  0 register dump:
PC      : 0x403836be  PS      : 0x00060036  A0      : 0x80381bb0  A1      : 0x3fcf0d00  
A2      : 0x3fc98dc4  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80383851  A9      : 0x3fc9b5f0  
A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x03c98cc0  A15     : 0x3fc98cc0  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x403836bb:0x3fcf0d00 0x40381bad:0x3fcf0d30 0x40383a48:0x3fcf0d60 0x40383a3e:0xa5a5a5a5 |<-CORRUPTED
```

Thanks in advance,

capc0307.

capc0307, 2024-06-17:
I've read the [issue #686](https://github.com/h2zero/NimBLE-Arduino/issues/686) called _NimBLE-Arduino crashes with esp32 version 3.0.1. and a LilyGo T-display S3_ and it seems I have a similar issue (if not the same). I  just uninstall and reinstalled the NimBLE-Arduino library in the Arduino IDE and I still got the same problem.

h2zero, 2024-06-17:
You need to download the current commit not the release to use the fix.

capc0307, 2024-06-17:
> You need to download the current commit not the release to use the fix.

I downloaded the current commit and it worked perfectly. Thank you so much.

closed 2024-06-17 by capc0307