h2zero/NimBLE-Arduino #690: BLE SERVICE and WiFi.scanNetworks not wokring simulteneously

demirschmid61, 2024-06-28:
Hello sir,

I am using BLE and WIfi both in one project, also trying to get WiFi available network list via BLE on app. If i am stating BLE service in the setup and if trying to get wifi list its showing WIFI_SCAN_FAILED. 

If the BLE is not initialized then wifi scan list is perfectly working.

I am using esp32-arduino-libs with version **3.0.0** and nimBLE version is **1.4.2**

h2zero, 2024-06-28:
Hello @demirschmid61 are you advertising on BLE at the same time as wifi scanning? Can you share a code snippet?

demirschmid61, 2024-06-30:
@h2zero Thank you very much for your response. Please go through my code snippet. I am using platformio for the development and using arduino framework. The platfnorm.ini file is also added, please have a look.

My observations:
1. If BLE is not initialized then l can see all the wifi list available and if BLE is not used the wifi list can be seen completely.

**Code Snippet**
```
#include <Arduino.h>
#include <FS.h>
#include "LittleFS.h"
#include <WiFi.h>
#include <NimBLEDevice.h>

static NimBLEServer *pServer;

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */
class ServerCallbacks : public NimBLEServerCallbacks
{
  void onConnect(NimBLEServer *pServer)
  {
    Serial.println("Client connected");
    Serial.println("Multi-connect support: start advertising");
    NimBLEDevice::startAdvertising();
  };

  void onConnect(NimBLEServer *pServer, ble_gap_conn_desc *desc)
  {
    Serial.print("Client address: ");
    Serial.println(NimBLEAddress(desc->peer_ota_addr).toString().c_str());
    pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
  };
  void onDisconnect(NimBLEServer *pServer)
  {
    Serial.println("Client disconnected - start advertising");
    NimBLEDevice::startAdvertising();
  };
  void onMTUChange(uint16_t MTU, ble_gap_conn_desc *desc)
  {
    Serial.printf("MTU updated: %u for connection ID: %u\n", MTU, desc->conn_handle);
  };

  /********************* Security handled here **********************
  ****** Note: these are the same return values as defaults ********/
  uint32_t onPassKeyRequest()
  {
    Serial.println("Server Passkey Request");
    return 123456;
  };

  bool onConfirmPIN(uint32_t pass_key)
  {
    Serial.print("The passkey YES/NO number: ");
    Serial.println(pass_key);
    /** Return false if passkeys don't match. */
    return true;
  };

  void onAuthenticationComplete(ble_gap_conn_desc *desc)
  {
    /** Check that encryption was successful, if not we disconnect the client */
    if (!desc->sec_state.encrypted)
    {
      NimBLEDevice::getServer()->disconnect(desc->conn_handle);
      Serial.println("Encrypt connection failed - disconnecting client");
      return;
    }
    Serial.println("Starting BLE work!");
  };
};
[... 247 lines cut ...]

```
[platformio]
default_envs = esp32-development-board

[env:esp32-development-board]
platform = espressif32 @ 6.6.0
board = esp-wrover-kit
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0
	framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.0/esp32-arduino-libs-3.0.0.zip
framework = arduino
lib_deps =
		lorol/LittleFS_esp32@1.0.5
		h2zero/NimBLE-Arduino @ 1.4.2

monitor_speed = 115200
    
board_build.filesystem = littlefs
board_upload.flash_size = 16MB
board_build.flash_mode = dio
board_build.partitions = default_16MB.csv
board_build.f_flash    = 80000000L
monitor_filters= time, log2file
```