h2zero/NimBLE-Arduino #692: GAP server event to get client rssi

igna09, 2024-07-05:
Hello,

i ported my project from bluedrod to NimBLE
thank you for your amazing work

In my project i need to get rssi of connected client

in bluedroid I used to make this:
```
esp_err_t rc = esp_ble_gap_read_rssi(this->authenticatedBdAddress->getNative());
...
customGapCallback(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
    switch(event) {
        ......
        case ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT: {
            String message = "[status: ";
            message += param->read_rssi_cmpl.status;
            message += ", rssi: ";
            message += param->read_rssi_cmpl.rssi;
            message += ", remote_addr: ";
            message += BLEAddress(param->read_rssi_cmpl.remote_addr).toString().c_str();
            message += "]";
...
            break;
        } // ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT
    }
}
```

In NimBLE I can't find a way to trigger event ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT to get rssi

h2zero, 2024-07-05:
You don't need to handle an event here for this. There is a low level function you can call: [int ble_gap_conn_rssi(uint16_t conn_handle, int8_t *out_rssi);](https://github.com/h2zero/NimBLE-Arduino/blob/44a69ed80518127e86302115bc6f985c285d5d2f/src/nimble/nimble/host/include/host/ble_gap.h#L2388)

I will look to add this to the CPP API in the future.

igna09, 2024-07-06:
it worked, thank you!

closed 2024-07-06 by igna09

lazd, 2025-05-25:
Thanks for this, here's what I ended up with to periodically perform RSSI checks using NimBLE and the the `ble_gap_conn_rssi` gap function:

```cpp
#define RSSI_INTERVAL         0.5

Ticker rssiTicker;
int16_t connectionHandle = -1;

void getRSSI() {
  int8_t rssi = 0;
  ble_gap_conn_rssi(connectionHandle, &rssi);

  Serial.print("📶 RSSI: ");
  Serial.print(rssi);
  Serial.println(" dBm");
}

void startRSSICheck() {
  rssiTicker.attach(RSSI_INTERVAL, getRSSI);
}

void stopRSSICheck() {
  rssiTicker.detach();
}

class ServiceServerCallbacks: public NimBLEServerCallbacks {
  void onConnect(NimBLEServer* server, NimBLEConnInfo& connInfo) override {
    Serial.println("🛜 Connected to device");
    connectionHandle = connInfo.getConnHandle();
  }
} serviceServerCallbacks;

// ... other fun stuff

void setup() {
  // ... other setup stuff
  NimBLEServer* server = NimBLEDevice::createServer();
  server->setCallbacks(&serviceServerCallbacks);

  startRSSICheck();
}
```