h2zero/NimBLE-Arduino #694: increasing heap consumption over time

Tschmidt9490, 2024-07-06:
My heap is slowly getting eaten up over time. This is ultimately causing a reboot of the device over time. I get the following issues. 

Backtrace: 0x40083845:0x3fff1d70 0x400940b1:0x3fff1d90 0x40099cd1:0x3fff1db0 0x4017a3fb:0x3fff1e30 0x4017a442:0x3fff1e50 0x4017b1b1:0x3fff1e70 0x4017a568:0x3fff1e90 0x4018abe1:0x3fff1eb0 0x4018f41e:0x3fff1ef0 0x4018f83c:0x3fff1f60 0x40194f26:0x3fff1f80 0x40194d0a:0x3fff1fc0 0x40195144:0x3fff1fe0 0x4019376d:0x3fff2000 0x40090a46:0x3fff2020 0x40188f2a:0x3fff2040

  #0  0x40083845:0x3fff1d70 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x400940b1:0x3fff1d90 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40099cd1:0x3fff1db0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4017a3fb:0x3fff1e30 in pbus_rx_dco_cal_1step at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:729
  #4  0x4017a442:0x3fff1e50 in pbus_rx_dco_cal_1step at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:738
  #5  0x4017b1b1:0x3fff1e70 in ram_rfcal_pwrctrl at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:1685
  #6  0x4017a568:0x3fff1e90 in pbus_rx_dco_cal_1step at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:769 (discriminator 1)
  #7  0x4018abe1:0x3fff1eb0 in NimBLEDevice::setSecurityAuth(unsigned char) at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/NimBLEDevice.cpp:1021
  #8  0x4018f41e:0x3fff1ef0 in ble_att_svr_fill_type_value at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att_svr.c:1159
  #9  0x4018f83c:0x3fff1f60 in ble_att_svr_prev_handle at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_att_svr.c:143
  #10 0x40194f26:0x3fff1f80 in ble_gatts_count_cfg at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gatts.c:2161
  #11 0x40194d0a:0x3fff1fc0 in ble_gatts_chr_updated at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gatts.c:1612
  #12 0x40195144:0x3fff1fe0 in ble_hs_timer_sched at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:482
  #13 0x4019376d:0x3fff2000 in ble_gattc_disc_chr_uuid_rx_adata at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:2487
  #14 0x40090a46:0x3fff2020 in ble_npl_event_run at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:526
      (inlined by) nimble_port_run at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:319
  #15 0x40188f2a:0x3fff2040 in std::enable_if<std::__and_<std::__not_<std::__is_tuple_like<unsigned char*> >, std::is_move_constructible<unsigned char*>, std::is_move_assignable<unsigned char*> >::value, void>::type std::swap<unsigned char*>(unsigned char*&, unsigned char*&) at c:\users\scoth\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/move.h:194
      (inlined by) std::_Vector_base<unsigned char, std::allocator<unsigned char> >::_Vector_impl::_M_swap_data(std::_Vector_base<unsigned char, std::allocator<unsigned char> >::_Vector_impl&) at c:\users\scoth\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_vector.h:112
      (inlined by) std::vector<unsigned char, std::allocator<unsigned char> >::swap(std::vector<unsigned char, std::allocator<unsigned char> >&) at c:\users\scoth\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_vector.h:1373
      (inlined by) NimBLEAdvertising::setManufacturerData(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at .pio/libdeps/adafruit_feather_esp32_v2/NimBLE-Arduino/src/NimBLEAdvertising.cpp:159
      
      
  Is there a way to track this down? 
  
  
  I believe it might be related to this function in my code 
  
  void BLEAppInterface_startAdvertising() {
    if (!pAdvertising) {
        pAdvertising = NimBLEDevice::getAdvertising();
    } else {
        pAdvertising->stop(); // Ensure previous advertising is stopped
    }

    if (!isUUIDAdded) {
        try {
            pAdvertising->addServiceUUID(pFireplaceControlService->getUUID());
            isUUIDAdded = true;
        } catch (const std::bad_alloc& e) {
            Serial.println("Failed to add Service UUID to Advertising: Out of Memory");
            return;
        } catch (...) {
            Serial.println("Failed to add Service UUID to Advertising: Unknown Error");
            return;
        }
    }

    uint8_t currentTick = _getCurrentTick();
    uint8_t currentStatus = _getCurrentStatus();
    uint8_t currentTankLevel = _getCurrentTankLevel();
    uint8_t currentFuelConsumption = _getCurrentFuelConsumption();

    // Check if there are changes in the values (excluding currentTick)
    if (currentStatus == previousStatus && 
        currentTankLevel == previousTankLevel && 
        currentFuelConsumption == previousFuelConsumption) {
        Serial.println("No changes in advertising data, skipping advertising.");
        return;
    }

    uint8_t Adv_DATA[] = {
        0x7f, 0x01, // build ManufacturerData
        0x7e, 0x88, 0xfd,
        currentTick,
        0xa4, 0xa5, 0xa6, 0xa7, 0xa8, 0xa9, 0xaa, 0xab,
        currentStatus, currentTankLevel, currentFuelConsumption, 0x00, 0x00, _getCurrentTankSize()
    };

    std::string manufacturerData(reinterpret_cast<char*>(Adv_DATA), sizeof(Adv_DATA));
    pAdvertising->setManufacturerData(manufacturerData); // Only one argument now
    pAdvertising->setScanResponse(true);
    pAdvertising->start();

    Serial.println("Advertising Started");

    // Update previous values
    previousStatus = currentStatus;
    previousTankLevel = currentTankLevel;
    previousFuelConsumption = currentFuelConsumption;
}

closed 2024-07-06 by Tschmidt9490

h2zero, 2024-07-06:
I'm assuming you found the issue? Was it something in the library?

Tschmidt9490, 2024-07-07:
I made a mistake with some connection for my BLE HID remote. It seems to be fixed and had nothing to do with the App implementation side of the code. The heap is now under control.

 I do have a reconnection issue though. I am almost giving up on getting this remote to reconnect after a reboot of the esp32...

h2zero, 2024-07-07:
Thanks. What is happening with reconnecting? Some changes were made lately to help reconnecting with bonded devices.

Tschmidt9490, 2024-07-07:
It does not seem to do it at all actually. Once i restart i need to have a button pressed on the remote. If not no device is shown which seems okay. I guess the remote is going in to sleep mode if no button is pressed for X time. 

I use a slightly modified version of the "NimBLE_client.ino" example to accomodate the HID spec. 

#include "remote.h"
#include <NimBLEDevice.h>
#include <ui.h>

static uint32_t scanTime = 0;  // Scan time in seconds

int bleServers = 0;
bool doConnect = false;
NimBLEAdvertisedDevice *advertisedDevice = nullptr;
bool commandProcessed = false;
char hexStr[5] = {0};
extern unsigned long lastUserActivityTime;

static NimBLEUUID batteryServiceUUID("180F");
static NimBLEUUID batteryLevelCharUUID("2A19");

extern bool remoteDebug;

void setHexStr(const char* value) {
    strncpy(hexStr, value, sizeof(hexStr) - 1);
    hexStr[sizeof(hexStr) - 1] = '\0'; // Ensure null termination
    commandProcessed = false; // Reset the processed flag when a new value is set
}

const char* getHexStr() {
    return hexStr;
}

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */
    void ClientCallbacks::onConnect(NimBLEClient* pClient) {
        Serial.println("Connected");
        /** After connection we should change the parameters if we don't need fast response times.
         *  These settings are 150ms interval, 0 latency, 450ms timout.
         *  Timeout should be a multiple of the interval, minimum is 100ms.
         *  I find a multiple of 3-5 * the interval works best for quick response/reconnect.
         *  Min interval: 120 * 1.25ms = 150, Max interval: 120 * 1.25ms = 150, 0 latency, 60 * 10ms = 600ms timeout
         */
        pClient->updateConnParams(120,120,0,60);
    }
    void ClientCallbacks::onDisconnect(NimBLEClient* pClient) {
        Serial.print(pClient->getPeerAddress().toString().c_str());
        Serial.println(" Disconnected - Starting scan");
        NimBLEDevice::getScan()->start(scanTime, scanEndedCB);
    }

    /** Called when the peripheral requests a change to the connection parameters.
     *  Return true to accept and apply them or false to reject and keep
     *  the currently used parameters. Default will return true.
     */
    bool ClientCallbacks::onConnParamsUpdateRequest(NimBLEClient* pClient, const ble_gap_upd_params* params) {
       /*
        if(params->itvl_min < 24) { 
            return false;
        } else if(params->itvl_max > 40) { 
            return false;
        } else if(params->latency > 2) { 
            return false;
        } else if(params->supervision_timeout > 100) { 
            return false;
        }
        */
        return true;
    }

    /** Pairing process complete, we can check the results in ble_gap_conn_desc */
[... 250 lines cut ...]

    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT); //use numeric comparison

    /** create new scan */
    NimBLEScan* pScan = NimBLEDevice::getScan();

    /** create a callback that gets called when advertisers are found */
    pScan->setAdvertisedDeviceCallbacks(new AdvertisedDeviceCallbacks());
    pScan->setInterval(45);
    pScan->setWindow(15);
    pScan->setActiveScan(true);
    pScan->start(scanTime, scanEndedCB);
}

void remoteTick() {
    if (doConnect) {
        doConnect = false;
        if (!connectToServer()) {
            if (remoteDebug) {
                Serial.println("failed to connect, starting scan");
            }
            //NimBLEDevice::getScan()->start(scanTime,scanEndedCB);
        }
    }
}

Tschmidt9490, 2024-07-08:
Now i have got it to reconnect. I had a variable declared at bool and not static bool which somehow made the difference. Yay.

Now i am keen to find a faster reconnect. It seems to need to rediscover all characteristics and resubscribe after a reboot. Is this needed? Should it save the device information for the next reconnect?

h2zero, 2024-07-08:
Great!

Yes, when reconnecting you should set the deleteAttributes parameter to false so you won't need to discover them again.

Tschmidt9490, 2024-07-08:
the one you are referring to is this one right?

        if (!pClient->connect(advertisedDevice,false)) {
            /** Created a client but failed to connect, don't need to keep it as it has no data */
            NimBLEDevice::deleteClient(pClient);
            Serial.println("Failed to connect, deleted client");
            resumeOtherTasks();
            return false;
        }

Tschmidt9490, 2024-07-08:
here is the whole function just for reference 

bool connectToServer() {
    suspendOtherTasks();
    lastUserActivityTime = millis();
    lv_label_set_text(ui_Label10, "Connecting remote");
    lv_obj_set_style_text_font(ui_Label10, &lv_font_montserrat_28, LV_PART_MAIN | LV_STATE_DEFAULT);
    int startTime = millis();
    NimBLEClient *pClient = nullptr;

    if (NimBLEDevice::getClientListSize()) {
        pClient = NimBLEDevice::getClientByPeerAddress(advertisedDevice->getAddress());
        if (pClient) {
            if (!pClient->connect(advertisedDevice, false)) {
                if (remoteDebug) {Serial.println("reconnect failed");}
                resumeOtherTasks();
                return false;
            }
            if (remoteDebug) {Serial.println("reconnected client");}
        } else { pClient = NimBLEDevice::getDisconnectedClient(); }
    }

    /** No client to reuse? Create a new one. */
    if(!pClient) {
        if(NimBLEDevice::getClientListSize() >= NIMBLE_MAX_CONNECTIONS) {
            Serial.println("Max clients reached - no more connections available");
            resumeOtherTasks();
            return false;
        }

        pClient = NimBLEDevice::createClient();

        Serial.println("New client created");

        pClient->setClientCallbacks(&clientCB, false);
        /** Set initial connection parameters: These settings are 15ms interval, 0 latency, 120ms timout.
         *  These settings are safe for 3 clients to connect reliably, can go faster if you have less
         *  connections. Timeout should be a multiple of the interval, minimum is 100ms.
         *  Min interval: 12 * 1.25ms = 15, Max interval: 12 * 1.25ms = 15, 0 latency, 51 * 10ms = 510ms timeout
         */
        pClient->setConnectionParams(6,12,0,100);
        /** Set how long we are willing to wait for the connection to complete (seconds), default is 30. */
        pClient->setConnectTimeout(10);

        if (!pClient->connect(advertisedDevice,false)) {
            /** Created a client but failed to connect, don't need to keep it as it has no data */
            NimBLEDevice::deleteClient(pClient);
            Serial.println("Failed to connect, deleted client");
            resumeOtherTasks();
            return false;
        }
    }

    if (!pClient->isConnected()) {
        if (!pClient->connect(advertisedDevice)) {
            if (remoteDebug) {
                Serial.println("failed to connect");
            }
            resumeOtherTasks();
            return false;
        }
    }

    // Use the discovered services and characteristics
    auto services = pClient->getServices(true);
    static NimBLEUUID hidServiceUUID(HID_SERVICE);
    for (auto service : *services) {
        suspendOtherTasks();
        auto serviceUUID = service->getUUID();
        if (serviceUUID.equals(hidServiceUUID)) {
[... 70 lines cut ...]
                    
                    // Subscribe to battery level notifications
                    if (characteristic->canNotify()) {
                        if (remoteDebug) {
                            Serial.println("-----> Subscribing to Battery Level Notifications...");
                        }
                        characteristic->subscribe(true, buttonPressCallback, true);
                    }
                }
                resumeOtherTasks();
            }
        }
        resumeOtherTasks();
    }

    char timeString[4];
    float seconds = (millis() - startTime) / 1000.0;
    snprintf(timeString, sizeof(timeString), "%3.2f", seconds);
    if (remoteDebug) {
        Serial.print("completed connection after ");
        Serial.println(timeString);
    }
    resumeOtherTasks();
    return true;
}

Tschmidt9490, 2024-07-08:
i can also see that this 

NimBLEDevice::getClientListSize()

Returns a 0 so it does not seem to be stored

Tschmidt9490, 2024-07-09:
i can see that after a successful connection is made with the connect to server that a bond if formed and the ClientListSize is increased by 1. The data is not persistent between reboots though

Tschmidt9490, 2024-07-09:
By pulling the battery of the remote and inserting again and not rebooting i can reconnect. But the code performs a full discovery again

h2zero, 2024-07-10:
The attributes are not stored in flash, only bond info is if enabled, so you would need to do discovery after reboot.