h2zero/NimBLE-Arduino #695: NimBLE-Arduino 1.4.2 is compatible with ArduinoESP32 3.0.2?

osmany-ferrer, 2024-07-07:
Hello,

My sketch setups NimBLEDevice::init("TEST"); but only "TE" name is advertised. Any name attempt only shows its first two letters.
This happened after updating NimBLE-Arduino to 1.4.2 and ArduinoEsp32 3.0.2 update was aborted.
My board is ArduinoEsp32 2.0.11
Multiple nrFConnect clients see the same two-letter name.

While updating ArduinoESP32 3.0.2 i suddently read this, so i aborted.

Migration from 2.x to 3.0
BLE
Changes in APIs
    Changed APIs return and parameter type from std::string to Arduino style String.
    Changed UUID data type from uint16_t to BLEUUID class.
    BLEScan::start and BLEScan::getResults methods return type changed from BLEScanResults to BLEScanResults*.
Is NimBLE-Arduino 1.4.2 compatible to ArduinoESP32 3.0.2?

Any idea, please, this is driving me crazy,
Thanks in advance,
Osmany

osmany-ferrer, 2024-07-07:
Solution,

I changed
   `pAdvertising->setScanResponse(false);`
to
     `pAdvertising->setScanResponse(true);`

h2zero, 2024-11-27:
Seems to be a common thing, the original library did not include the flags data in the advertisement, which allowed for 3 more bytes to be advertised. You could do this here as well instead of turning on the scan response by using custom data that does not include the flags, but I don't recommend it as it goes against the BLE specs unless your device is not connectable.

closed 2024-12-02 by h2zero