h2zero/NimBLE-Arduino #705: Cannot set TX power to +8dbm for nRF52840

tnicolas42, 2024-08-06:
I would like to use an nRF52840 chip as a NimBLEDevice with a TX power of +8 dBm. The problem is that when I call NimBLEDevice::setPower(8);, it automatically caps at 4 (due to the function below).

How can I resolve this issue ?
Would simply adding this be sufficient ?
```c++
if (dbm >= (int8_t)RADIO_TXPOWER_TXPOWER_Pos8dBm) {
    return (int8_t)RADIO_TXPOWER_TXPOWER_Pos8dBm;
}
```

https://github.com/h2zero/NimBLE-Arduino/blob/6e1ad0fb5776e5023e1dae904d80ea60d943f872/src/nimble/nimble/drivers/nrf52/src/ble_phy.c#L1847-L1854

closed 2024-08-08 by tnicolas42

h2zero, 2024-08-08:
Hello @tnicolas42, looks like you found the solution, this has also been addressed in the NimBLE-Arduino master branch.

tnicolas42, 2024-08-09:
Hi,

I haven't found a solution yet, but I noticed some commits related to the 8 dBm power. I'm waiting for the next release to see if it resolves the issue.

h2zero, 2024-08-10:
Sounds good