h2zero/NimBLE-Arduino #708: NRA connection on NON-esp device (NRF81522)

VincentGijsen, 2024-08-11:
hi, 

i'm trying to connect as client to my hometrainer, alas so far i've not managed to do so.
presently using an NRF58122, where it seems this function() is non-existing: `void NimBLEDevice::setOwnAddrType(uint8_t own_addr_type, bool useNRPA) `. however i'm not entirely sure its even the right function in the first place.

While testing on an ESP32S3, i have this function, but after exhausting most combinations of that function also failed to connect to the service, i just get (in both cases) a connection back, but then getting a REF to a BLE-service, fails

using the the ANOTHER libary (#include "BLEDevice.h"), i was able to connect to this hometrainer, using this call: 
```// Connect to the remove BLE Server.
    bool _connected = pClient->connect(*myBike, BLE_ADDR_TYPE_RANDOM); //, BLE_ADDR_TYPE_PUBLIC);
    // the RANDOm bit is important, else it will not connect!
``` do note the explicit need to make clear the connect requires the BLE_ADDR_TYPE_RANDOM const, which is fed into the GATT object of the ESP-libary underlying.

hence I'm not entirely sure if the connect-argument of the stock-library, is 'similar'  as the `NimBLEDevice::setOwnAddrType(uint8_t own_addr_type, bool useNRPA) `

Any pointers how how i could get the connection going?
ideally I stick to the NRF with the great `https://github.com/h2zero/platform-n-able.git@^1.1.0` library. as these chips are soo dirt cheap, and i'm really charmed by the full open-source setup possible here (no softdevice) :)

ultimate goal would be to have the NRF/ESP as client+server to rewrite some calls between trainer and application on the pc albeit challenging given the very limited ram/flash on the NRF.
Debugging with the NRF is quite a bit faster/more pleasant compared to the ESP, with my Segger probe.

Thanks for the awesome work already; regardless of a solution for this one!

h2zero, 2024-08-29:
Hi, I'm not sure what the issue is in this case as the nRF chips naturally have a random static address, so I would think it should work. That said the device you're connecting to might be expecting only a phone to connect which will have a random resovlable address, requiring bonding. In this case there would need to be some changes to the library to support this as it's not currently implemented. I will flag this to investigate further.

VincentGijsen, 2024-08-30:
mm, sound not super trivial then?

anything i can do to 'log/dump' /handshake the bonding?

h2zero, 2024-09-15:
Try this:
```
ble_addr_t addr;
ble_hs_id_gen_rnd(0, &addr);
ble_hs_id_set_rnd(addr.val);
```

VincentGijsen, 2024-09-18:
i tried it, 

in my init function, around the init of nimbestack, and (seperately just before the connect function to the scanned/found MAC, for my trainer:

btw, the trainer reports: `- Resolvable Private Address Only [R] (0x2AC9) ` (NRFconnect); attached the full log; perhaps it tells you more then it tells mine.
[Tunturi E35 log NRFconnect.txt](https://github.com/user-attachments/files/17040782/Tunturi.E35.log.NRFconnect.txt)

```
    ble_addr_t addr;
    ble_hs_id_gen_rnd(0, &addr);
    ble_hs_id_set_rnd(addr.val);

    this->_pClient = NimBLEDevice::createClient();
    this->_pClient->setClientCallbacks(&clientCB, false);

//ble_addr_t addr;
    ble_hs_id_gen_rnd(0, &addr);
    ble_hs_id_set_rnd(addr.val);
<<<<LOG BELOW>>>
D NimBLEClient: Got Client event 
E NimBLEClient: Connection failed; status=13
Failed to connect
..
```
like so:
```
..
bool BikeClient::connectToTrainer()
{
    log_i("starting");
  
    
    ble_addr_t addr;
    ble_hs_id_gen_rnd(0, &addr);
    ble_hs_id_set_rnd(addr.val);

    doConnect = false;

        if (!_pClient->isConnected())
        {
            if (!_pClient->connect(advDevice,true))
            {
                Serial.println("Failed to connect");
                return false;

...

```

not sure if these are the right place to call those functions, irrespectively both flavors compile fine.

h2zero, 2024-09-21:
As long as you set the address after `init` it should be fine. What is the error code when the connection fails?

VincentGijsen, 2024-09-22:
Could it be‘’NimBLEClient: Connection failed; status=13’’’Or some lower level code I’m not printing

h2zero, 2024-11-20:
Sorry, I forgot to get back to this. That is a timeout error, the device didn't respond to the connection request in time. Have you made any progress?