h2zero/NimBLE-Arduino #711: Adding - removing characteristic on runtime issue

sanastasiou, 2024-08-19:
Hi @h2zero , thanks a lot for your library.

I came to notice an issue, while adding a characteristic / removing it on runtime.

While the service appears to be updated, the advertisement data is not.

Here is a small example:

`        // setup non encrypted characteristics
        fPServiceCharacteristics[CHARACTERISTICS_INDEX::NON_ENCRYPTED] = addCharacteristicToService(constants::NON_ENCRYPTED_CHARACTERISTIC_UUID.c_str(), NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
        fPServiceCharacteristics[CHARACTERISTICS_INDEX::NON_ENCRYPTED]->setCallbacks(new ble::PingCallbacks());

        //start service
        fPKillswitchService->start();

        fPServer->getAdvertising()->addServiceUUID(constants::SERVICE_UUID.c_str());
        fPServer->getAdvertising()->setAppearance(576U);
        fPServer->getAdvertising()->setScanResponse(true);

    NimBLECharacteristic *BluetoothController::addCharacteristicToService(const char *uuid, const uint32_t properties)const
    {
        NimBLECharacteristic *pCharacteristic = fPKillswitchService->createCharacteristic(
            uuid,
            properties);
        return pCharacteristic;
    }

    //later... 
    void BluetoothController::addEncryptedCharacteristics()
    {
        if(!fEncryptedServicesStarted)
        {
            while(false == NimBLEDevice::stopAdvertising());
            NimBLEDevice::getAdvertising()->removeServices();
            //adding encrypted characteristics
            fPServiceCharacteristics[CHARACTERISTICS_INDEX::ENCRYPTED] = addCharacteristicToService(constants::ENCRYPTED_CHARACTERISTIC_UUID.c_str(), NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::READ_AUTHEN | NIMBLE_PROPERTY::NOTIFY | NIMBLE_PROPERTY::INDICATE | NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_AUTHEN | NIMBLE_PROPERTY::WRITE_ENC);
            fPServiceCharacteristics[CHARACTERISTICS_INDEX::ENCRYPTED]->setCallbacks(new ble::BatteryLockCallbacks(fBatteryLocked));
            fPServiceCharacteristics[CHARACTERISTICS_INDEX::ENCRYPTED]->setValue(fBatteryLocked);
            const auto v = fPKillswitchService->getCharacteristics();
            for(auto c: v)
            {
                std::cout << "Characteristics in service: [" << c->getUUID().toString() << "]" << std::endl;
            }
            std::cout << "Transmitting encrypted characteristics started." << std::endl;
            fEncryptedServicesStarted = true;
            NimBLEDevice::getAdvertising()->addServiceUUID(constants::SERVICE_UUID.c_str());
            while(false == NimBLEDevice::startAdvertising());
        }
    }

> D NimBLEService: >> start(): Starting service: UUID: 00002ffa-0000-1000-8000-00805f9b34fb, handle: 0xffff
D NimBLEService: Adding 1 characteristics for service UUID: 00002ffa-0000-1000-8000-00805f9b34fb, handle: 0xffff
D NimBLEService: << start()
Bluetooth initialization done...
FSM -> 0 -> Init().
Initializing display controller
SSD1306 initialized properly...
DeferredAdController::start()
DeferredAdController::start() - fTransitionTimeout.reset()
Number of bonded devices: 0
FSM -> Init() -> Pair().
Real voltage:      17.41 V
Is advertising: false
Is connected: false
Is bonded: false
Ping value: 0
Encrypted communications started: false
DeferredAdController::trace() : State: 1 - Previous State: 0 - transition timer: 0
LedControl::update() - state: Init
D NimBLEAdvertising: >> stop
D NimBLEAdvertising: << stop
Characteristics in service: [82c314b2-e53a-4d5e-b021-fc5887b4373f]
Characteristics in service: [3c5577d7-d182-482a-940a-7e5062441f43]
Transmitting encrypted characteristics started.
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 00002ffa-0000-1000-8000-00805f9b34fb
         handle 10
     end_handle 13
characteristic
           uuid 82c314b2-e53a-4d5e-b021-fc5887b4373f
     def_handle 11
     val_handle 12
   min_key_size 0
          flags [READ|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 13
   min_key_size 0
          flags [READ|WRITE]
D NimBLEAdvertising: << Advertising start

So the service, has indeed the two characteristics added, but the advertisement data does not.

Do you know what the proper way to add/remove a characteristic during runtime is?

Best Regards!

sanastasiou, 2024-08-19:
Is it possible that this is not implemented?

[image]

h2zero, 2024-08-19:
There is no provision the the BLE spec to advertise characteristics, only services.

sanastasiou, 2024-08-19:
@h2zero  thanks for replying, but even if I connect to the actual service although it contains both characteristics only one is there.

        const auto v = fPKillswitchService->getCharacteristics();
        
        for(auto c: v)
        {
            std::cout << "Characteristics in service: [" << c->getUUID().toString() << "]" << std::endl;
        }

The above code prints correctly both characteristics, but the clienst only sees one. Could u explain that?

h2zero, 2024-08-19:
You need to call `service::start()` after adding the characteristics. Please make sure there are no devices connected when you do this.

sanastasiou, 2024-08-19:
You mean disconnect all clients, and call start again although I have already done that ? And same when removing a characteristic?

h2zero, 2024-08-19:
Yes, the attributes cannot be changed while there are active connections because the handles need to be changed.

sanastasiou, 2024-09-10:
Hi @h2zero , after following your advice and implementing against it I can now verify that adding and removing characteristics dynamically works perfectly fine.

Many thanks to you sir!

sanastasiou, 2024-09-12:
Hi @h2zero , unfortunately I need to reopen this issue. While the adding/removing characteristics is now properly displayed in the service, the characteristics themselves stop working after one is removed.

My code service is pretty simple. It concists of one characteristic which is always present and never removed. And an encrypted characteristic which is added or removed on the fly. Following thing happens:

- Add non encryped characteristic : Check read/notify -> works
- Disconnect all devices
- Add encryped characteristics
- Reconnect device.
- Bond device.
- Read/Write/Notify from encryped characteristc -> works.
- Read/Notify from non encryped characteristic -> works.
- Disconnect all devices.
- Remove encryped characteristic.
- Reconnect previously bonded device.
- Disconnect device again and add encryped characteristic again.
- Reconnect previously bonded device.
- Read/Notify from non encryped characteristic no longer works ( no characteristic callbacks are called, valuea are no longer updated )
- Read/Notify/Write from encryped characteristic no longer works. No callbacks are called. Vaues are never updated.

Basically, the moment I call service->removeCharacteristic(...) while no device is connected, everything breaks.

What I tried:

Removing characteristic with the optional boolen to delete it with true/false. No difference.
Calling service->start() after removing the characteristic. No difference.
Not calling service->start() after removing the characteristic. No difference.

Would you happen to have any ideas as to what I might be doing wrong here?

` 

    bool ServiceHandler::removeEncrypedCharacteristics()
    {
        if(fPServer->getConnectedCount() == 0)
        {
            fPKillswitchService->removeCharacteristic(fPServiceCharacteristics[CHARACTERISTICS_INDEX::ENCRYPTED], true);
            //fPKillswitchService->start();
        }
        const auto v = fPKillswitchService->getCharacteristics();
        for(auto c: v)
        {
            std::cout << "ServiceHandler::removeEncrypedCharacteristics() - Characteristics in service: [" << c->getUUID().toString() << "]" << std::endl;
            if(0 == strcmp(c->getUUID().toString().c_str(), constants::ENCRYPTED_CHARACTERISTIC_UUID.c_str()))
            {
                return false;
            }
        }
        return true;
    }
`

sanastasiou, 2024-09-12:
Tried even removing the service itself. Did not help either. Only destroying the reinitializing the entire stack works:

```
        fPKillswitchService->removeCharacteristic(fPServiceCharacteristics[CHARACTERISTICS_INDEX::ENCRYPTED], true);
        fPServer->removeService(fPKillswitchService, true);
        NimBLEDevice::deinit(true);
        NimBLEDevice::init(constants::DEVICE_NAME);
        NimBLEDevice::setPower(ESP_PWR_LVL_P9);
        fPServer = NimBLEDevice::createServer();
        fPServer->setCallbacks(new ServerCallbacks(fDevicePin, fBleController));

        fPKillswitchService = fPServer->createService(constants::SERVICE_UUID.c_str());

        // setup non encrypted characteristics
        fPServiceCharacteristics[CHARACTERISTICS_INDEX::NON_ENCRYPTED] = fPKillswitchService->createCharacteristic(constants::NON_ENCRYPTED_CHARACTERISTIC_UUID.c_str(), NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY | NIMBLE_PROPERTY::INDICATE);
        fPServiceCharacteristics[CHARACTERISTICS_INDEX::NON_ENCRYPTED]->setCallbacks(new ble::PingCallbacks());

        //start service
        fPKillswitchService->start();

        fPServer->getAdvertising()->addServiceUUID(constants::SERVICE_UUID.c_str());
        fPServer->getAdvertising()->setAppearance(576U);
        fPServer->getAdvertising()->setScanResponse(true);
        fPServer->getAdvertising()->start();
    
```

After this, the first characteristic still works fine, callbacks, subscription etc. work. For me clearly some issue here but I can live with this workaround.

h2zero, 2024-09-15:
Thanks, seems there may be a bug in this case. I will need to look into this.

sanastasiou, 2024-09-15:
@h2zero no problem, would be happy to test any bug fix you might have.

h2zero, 2024-09-15:
Could you post some logs at debug level? That would help a lot.

sanastasiou, 2024-09-20:
Hi, sorry for replying late. WIll try to post the debug output today!

h2zero, 2024-10-11:
Thanks, looks like an android issue, or app issue there. The client device should subscribe to the service changed indication and request the database from the peripheral when it receives this indication, doesn't look like android is doing that.

sanastasiou, 2024-10-11:
The thing is however, the client has to disconnect in order to add the new characteristic right? If it does that, how should it subscribe to this event?

The way ESP32 is implemented right now is:

- Wait for client to connect
- Check if client is bonded
- If client is bonded and encryped characteristic is not added to the service, disconnect client, add it and wait for client to reconnect
- Client reconnects and sees both characteristics

Exactly this works in iOS but not on Android.

Android used to work, when I was adding the encrypted characteristic without disconnecting it first. So quite confused now.

h2zero, 2024-11-27:
@sanastasiou There might be some improvement to this in #763 if you'd like to test.

vandy, 2025-01-30:
@sanastasiou, also check whether it's an issue with Android caching services and characteristics. I've met in different sources that once connected Android caches information about services/characteristics and doesn't perform service/characteristics discovery next time. This could be the case. It's said that it's enough to switch off the Bluetooth to reset the cache (some said that the device should reboot).