h2zero/NimBLE-Arduino #713: Connecting to same device as both client & server for ANCS

mathcampbell, 2024-08-27:
hey there; I'm trying to implement an ANCS (Apple Notification Centre Service) using NimBLE - I'm having real trouble though because it needs you to make a server on your device, solicit the UUID for ANCS, connect the iOS device ("phone" to save me typing device constantly) as a client, authenticate etc, then create a client and connect the phone as a server, at the same time.

I have built my code virtually mirroring an implementation of ANCS for arduino built on bluedroid, but as soon as I try to connect the client it spits out "E NimBLEClient: A connection to xx:xx:xx:xx:xx:xx already exists" (and there isn't a client connection; it's the NimBLEServer connection to the phone).
I know the hardware can manage it etc; anyone care to help me out where I'm going wrong?

h2zero, 2024-08-29:
Hi, the master branch has added this capability by creating a client and then assigning the connection handle to it. 
See commit: 6db0368

h2zero, 2024-12-02:
@mathcampbell  A new PR #774 was opened to resolve this if you would like to test it. All feedback is greatly appreciated!

PR #774 (2024-12-05): Add support for creating a NimBLEClient from a NimBLEServer peer.
This allows the NimBLEServer instance to create a NimBLEClient instance to read/write form/to a connected peer. Only one instance is supported subsequent calls will overwrite the previous client connection information and data.

Fixes #713
closed 2024-12-05 by h2zero