h2zero/NimBLE-Arduino #715: Compile fails if using usb libraries because of HIDTypes.h:

psxde, 2024-08-30:
Hi, 

if I use NimBLE instead of the standard BLE stack together with USB Host, I get compile errors like that:

```
Compiling .pio\build\esp32-s3-devkitc-1\lib3e1\NimBLE-Arduino\NimBLEClient.cpp.o
In file included from .pio/libdeps/esp32-s3-devkitc-1/EspUsbHost/src/EspUsbHost.h:7,
                 from src/main.cpp:5:
.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_rom/include/esp32s3/rom/usb/usb_common.h:104: warning: "HID_CLASS" redefined
 #define HID_CLASS           0x03

In file included from .pio/libdeps/esp32-s3-devkitc-1/NimBLE-Arduino/src/NimBLEHIDDevice.h:24,
                 from .pio/libdeps/esp32-s3-devkitc-1/ESP32 BLE Keyboard/BleKeyboard.h:12,
                 from src/main.cpp:2:
.pio/libdeps/esp32-s3-devkitc-1/NimBLE-Arduino/src/HIDTypes.h:28: note: this is the location of the previous definition
 #define HID_CLASS           (3)

.pio/libdeps/esp32-s3-devkitc-1/NimBLE-Arduino/src/HIDTypes.h:29:29: error: expected identifier before '(' token
 #define HID_SUBCLASS_NONE   (0)
                             ^
```

h2zero, 2024-09-15:
Thanks, I will look to change these.

davidandreoletti, 2024-11-19:
@h2zero Using platformio and the following *versionned library deps* (with USB Host), the same build failure appears.

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/seeed_xiao_esp32s3.html
PLATFORM: Espressif 32 (6.9.0) > Seeed Studio XIAO ESP32S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-BLE-Gamepad @ 0.5.5
|-- AdvancedLogger @ 1.3.0
|-- Adafruit TinyUSB Library @ 3.4.0
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio/build/adapter_generic/src/USBHost.cpp.o
Building .pio/build/adapter_generic/bootloader.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generating partitions .pio/build/adapter_generic/partitions.bin
Compiling .pio/build/adapter_generic/lib5ee/NimBLE-Arduino/NimBLE2904.cpp.o
In file included from .pio/libdeps/adapter_generic/NimBLE-Arduino/src/NimBLEHIDDevice.h:24,
                 from .pio/libdeps/adapter_generic/ESP32-BLE-Gamepad/BleGamepad.h:10,
                 from src/main.cpp:12:
.pio/libdeps/adapter_generic/NimBLE-Arduino/src/HIDTypes.h:29:29: error: expected identifier before '(' token
 #define HID_SUBCLASS_NONE   (0)
```

h2zero, 2024-11-19:
@davidandreoletti If you could please check that this is resolved by #757 I will merge it

davidandreoletti, 2024-11-26:
@h2zero The compilation failure above with Both NimBLE-Arduino and Adafruint TinyUSB in the same translation unit is now gone with #757.

PR #757 (2024-11-26): Fix compilation errors in some situations.
Fixes #715
closed 2024-11-26 by h2zero