h2zero/NimBLE-Arduino #722: how do i add Descriptor to my code?

Volanaro, 2024-09-29:
hey so iv made a 2 button gamepad, and im trying to get it to show up as Sequential Shifter in windows Game Controllers and not 8 axis 2 button device, iv tired loads of diffrent code, and iv tired looking in the header files for help.... how do i do this? this is my code:

`#include <BleGamepad.h>
#include <BleGamepadConfiguration.h>

// Pin Definitions
const int BUTTON_1_PIN = 12;  // Pin for Button 1
const int BUTTON_2_PIN = 13;  // Pin for Button 2

// Create a gamepad configuration
BleGamepadConfiguration config;

// Initialize the gamepad instance with the name 'Gamepad Shifter'
BleGamepad gamepad("Sequential Shifter", "Craig Jenkins", 100);

void setup() {
    Serial.begin(115200);

    // Setup the gamepad configuration explicitly
    config.setModelNumber("Sequential Shifter"); 
    config.setSoftwareRevision("1.0");
    config.setSerialNumber("123456789");
    config.setFirmwareRevision("1.0");
    config.setHardwareRevision("1.0");
    
    config.setVid(0x1234);  // Set your Vendor ID
    config.setPid(0x5678);  // Set your Product ID
    config.setGuidVersion(0x0110); // Set your GUID Version

    // Only declare buttons
    config.setButtonCount(2); // Set button count to 2
    config.setHatSwitchCount(0); // No hat switches

    gamepad.begin(&config); // Initialize with the configuration

    // Setup button pins as input with pull-ups
    pinMode(BUTTON_1_PIN, INPUT_PULLUP);
    pinMode(BUTTON_2_PIN, INPUT_PULLUP);
}

void loop() {
    static bool button1State = false;
    static bool button2State = false;

    bool currentButton1State = digitalRead(BUTTON_1_PIN) == LOW;  // Low means pressed
    bool currentButton2State = digitalRead(BUTTON_2_PIN) == LOW;  // Low means pressed

    // Button 1 Logic
    if (currentButton1State != button1State) {
        button1State = currentButton1State;

        if (button1State) {
            gamepad.press(1); // Press Button 1
            Serial.println("Button 1 Pressed");
        } else {
            gamepad.release(1); // Release Button 1
            Serial.println("Button 1 Released");
        }
    }

    // Button 2 Logic
    if (currentButton2State != button2State) {
        button2State = currentButton2State;

        if (button2State) {
            gamepad.press(2); // Press Button 2
            Serial.println("Button 2 Pressed");
        } else {
            gamepad.release(2); // Release Button 2
            Serial.println("Button 2 Released");
        }
    }

    // Send updated gamepad state
    gamepad.sendReport();

    // Add a delay to debounce button presses
    delay(10);
}`

maxr217, 2024-12-09:
*Following*

I am making a single axis joystick controller to interact with Steam however regardless of code changes Steam still reads an "8 axis 16 button device with hat switch"

h2zero, 2024-12-09:
This has been covered in #703