h2zero/NimBLE-Arduino #724: Crash deinit

AlexSmirnov9107, 2024-10-01:
NimBle installed from zip from master branch

```
        BLEDevice::init(deviceName);
	BLEDevice::setSecurityAuth(true, true, true); // Disable bonding, MITM, and secure connections
	BLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
	// BLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
	// BLE_HS_IO_DISPLAY_ONLY
	// BLEDevice::setSecurityPasskey(123456);

	pServer = BLEDevice::createServer();
	pServer->setCallbacks(this);

	hid = new BLEHIDDevice(pServer);
	inputKeyboard = hid->inputReport(KEYBOARD_ID); // <-- input REPORTID from report map
	outputKeyboard = hid->outputReport(KEYBOARD_ID);
	inputMediaKeys = hid->inputReport(MEDIA_KEYS_ID);

	hid->manufacturer()->setValue(deviceManufacturer);

	hid->pnp(0x02, 0xe502, 0xa111, 0x0210);
	hid->hidInfo(0x00, 0x01);

	hid->reportMap((uint8_t *)_hidReportDescriptor, sizeof(_hidReportDescriptor));
	hid->setBatteryLevel(batteryLevel);
	hid->startServices();
	pCustomService = pServer->createService(CUSTOM_SERVICE_UUID);
	pCustomCharacteristic = pCustomService->createCharacteristic(
		CUSTOM_CHARACTERISTIC_UUID,
		NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::WRITE_ENC);
	pCustomService->start();

	onStarted(pServer);

	advertising = pServer->getAdvertising();
	advertising->setAppearance(HID_KEYBOARD);
	advertising->addServiceUUID(hid->hidService()->getUUID());
	advertising->setScanResponse(false);
	advertising->addServiceUUID(CUSTOM_SERVICE_UUID);
	advertising->start();
```
Caliing deinit produce crash

```
13:45:51.332 -> 
13:45:51.332 -> 
13:45:51.332 -> Backtrace: 0xfffffffd:0x3ffc86f0 0x400d573c:0x3ffc8710 0x400d4e27:0x3ffc8730 0x400d30fe:0x3ffc8750 0x400d2ce9:0x3ffc8770 0x400e85e4:0x3ffc87d0 0x40095cee:0x3ffc87f0
13:45:51.332 -> 
13:45:51.332 -> 
13:45:51.332 -> 
13:45:51.332 -> 
13:45:51.332 -> ELF file SHA256: 1e895c879a3550a0
13:45:51.332 -> 
13:45:51.462 -> Rebooting...
13:45:51.462 -> ets Jul 29 2019 12:21:46
13:45:51.462 -> 
13:45:51.462 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:45:51.494 -> configsip: 0, SPIWP:0xee
13:45:51.494 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:45:51.494 -> mode:DIO, clock div:1
13:45:51.494 -> load:0x3fff0030,len:4604
13:45:51.494 -> ho 0 tail 12 room 4
13:45:51.494 -> load:0x40078000,len:15488
13:45:51.494 -> load:0x40080400,len:4
13:45:51.494 -> load:0x40080404,len:3180
13:45:51.494 -> entry 0x400805b8
```

h2zero, 2024-10-02:
Thanks for the report, could you provide more logs and if possible a decode of the backtrace?

rac146, 2024-10-23:
Just saw a similar issue - you need to disconnect and stop advertising if you're going to pass true to deinit now. Also, make sure you wait for the disconnect callback to complete before calling `deinit(true)`.

 @h2zero the old way (code after nimble calls) would force a disconnection (via nimble_port_stop) before it disposed the `m_pServer` object.

h2zero, 2024-10-23:
Thank you for the feedback, I will work on a fix shortly.

PR #738 (2024-11-03): [BREAKING] Refactor NimBLEDevice
* General code cleanup
* `NimBLEDevice::getInitialized` renamed to `NimBLEDevice::isInitialized`.
* `NimBLEDevice::setPower` no longer takes the `esp_power_level_t` and `esp_ble_power_type_t`, instead only an integer value in dbm units is accepted.
* `NimBLEDevice::setPower` now returns a bool value, true = success.
* `NimBLEDevice::setMTU` now returns a bool value, true = success.
* `NimBLEDevice::injectConfirmPIN` renamed to `NimBLEDevice::injectConfirmPasskey` to use Bluetooth naming.
* Fixes crash if `NimBLEDevice::deinit` is called when the stack has not b [... cut]
closed 2024-11-03 by h2zero