h2zero/NimBLE-Arduino #726: Possible error in NimBLEAdvertisedDevice constructor

gkoh, 2024-10-04:
If `CONFIG_BT_NIMBLE_EXT_ADV` is not defined, the initialiser list for `NimBLEAdvertisedDevice` looks like this for `m_payload` (in `src/NimBLEAdvertisedDevice.cpp`):
```
m_payload{event->disc.data, event->disc.data + event->disc.length_data}
```
as `m_payload` is a `std::vector` this is valid a initialiser.

However, I think it should actually be:
```
m_payload(event->disc.data, event->disc.data + event->disc.length_data)
```
which matches the initialiser for `CONFIG_BT_NIMBLE_EXT_ADV=1`

I think this partially explains some weird behaviour I'm getting if I disable exceptions (via `-fno-exceptions`) in my esp32 project.
Am I crazy?

gkoh, 2024-10-04:
This is on master `23eb4539194fea88fa9cb465949b397efe04b231`, here:
https://github.com/h2zero/NimBLE-Arduino/blob/23eb4539194fea88fa9cb465949b397efe04b231/src/NimBLEAdvertisedDevice.cpp#L50

h2zero, 2024-10-06:
Thanks!, You're correct, I will address this right away!

gkoh, 2024-10-12:
> @gkoh Please check the latest commit and let me know if this is resolved.

Looks good in a79941c, thank you.

closed 2024-10-12 by gkoh