h2zero/NimBLE-Arduino #727: OnAuthenticationComplete callback - NimBLEDevice::isBonded function behavior.

sanastasiou, 2024-10-04:
```
        void ServerCallbacks::onAuthenticationComplete(ble_gap_conn_desc *desc)
        {
            const std::lock_guard<std::mutex> guard(fBleController.getMutex());
            const bool isBonded = (desc->sec_state.bonded == 1);
            const bool isEncrypted = (desc->sec_state.encrypted == 1);
            const bool isAuthenticated = (desc->sec_state.authenticated == 1);
            std::cout << std::boolalpha << "ServerCallbacks::onAuthenticationComplete() - encrypted: " << isEncrypted << std::endl;
            std::cout << std::boolalpha << "ServerCallbacks::onAuthenticationComplete() - authenticated: " << isAuthenticated << std::endl;
            std::cout << std::boolalpha << "ServerCallbacks::onAuthenticationComplete() - device address type: " << (int)desc->peer_id_addr.type << std::endl;
            std::cout << std::boolalpha << "ServerCallbacks::onAuthenticationComplete() - bonded: " << isBonded << std::endl;
            std::cout << std::boolalpha << "ServerCallbacks::onAuthenticationComplete() - bonded: " << NimBLEDevice::isBonded(NimBLEAddress(desc->peer_id_addr)) << std::endl;
            ...
        }
```

During authentication / bonding this function will print:

```
ServerCallbacks::onAuthenticationComplete() - encrypted: true
ServerCallbacks::onAuthenticationComplete() - authenticated: true
ServerCallbacks::onAuthenticationComplete() - device address type: 1
ServerCallbacks::onAuthenticationComplete() - fBleController.isPairing(): true
ServerCallbacks::onAuthenticationComplete() - bonded: true
ServerCallbacks::onAuthenticationComplete() - bonded: false
```

In the meantime, all encrypted communications/calbacks work properly. Even after the callback has finished, this function returns false.

_After disconnection and new connection_ with the same device the function prints `NimBLEDevice::isBonded(NimBLEAddress(desc->peer_id_addr))` as true.

Is this the expected behavior?

sanastasiou, 2024-10-04:
Update: this also works properly:

`std::cout << std::boolalpha << "Is bonded check via server info: " << NimBLEDevice::getServer()->getPeerInfo(NimBLEDevice::getServer()->getPeerDevices()[0]).isBonded() << std::endl;`

Will report true during and after authentication.

h2zero, 2024-10-06:
This is expect behavior unfortunately, its just the wat the stack responds to the function call after the bonding event.

sanastasiou, 2024-10-06:
@h2zero thanks. Closing this as a non issue then.

closed 2024-10-06 by sanastasiou