h2zero/NimBLE-Arduino #739: New User Guide Create a Server example not visible on Windows PC.

YFL, 2024-11-02:
Hey, I'm trying to create an XInput Controller and ATM I'm at the point where I'm just trying to connect the EFS32 to the PC with NimBLE.

### The problem
I tried using the basic example from the [New User Guide](https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/New_user_guide.md#creating-a-server), but I don't see the device in the Windows 11 Settings.

### Notes

- I see the device on my Android phone both in the settings and in nRF connect.
- I'm using a Bluetooth Dongle (no BT 5 support) on the PC, which I've confirmed, that works with other devices.
- I've tried using other libraries (like [BLE-Gamepad](https://github.com/lemmingDev/ESP32-BLE-Gamepad) or [BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID)), which use NimBLE-Arduino under the hood, and I was seeing the ESP32 on Windows, but it just keeps disconnecting (seems like the connection times out, not sure, if Windows closes the connection, or something else's happening).

h2zero, 2024-11-03:
Not sure what the problem could be, do you have another PC to compare with?

YFL, 2024-11-09:
Okay, so I tried a different PC and I can't see it there either.

h2zero, 2024-11-09:
Sorry, I misunderstood your issue. I thought you already created and hid device and it wasn't showing in windows. 

Windows will not show devices that are not devices that it knows how to talk to, hid, audio, etc.. 

The example in the docs does not create such a device as that is a specific application of the library. What you will see with that example is it will show up in a scan with a phone app such as nRFconnect.

YFL, 2024-11-09:
Ah, okay, understood. Thank you. That works. I'll close this one then :)

closed 2024-11-09 by YFL