h2zero/NimBLE-Arduino #745: Build errors in VScode with Platformio

BasvanH, 2024-11-04:
Hi,

Im unable to build NimBLE in VScode with PlatformIO.
Windows 11
PlatformIO: 6.1.16
VScode: 1.95.0

This is my platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages = 
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#1.0.6
lib_deps = h2zero/NimBLE-Arduino@1.4.2
```

And this is the build log;
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2024.7.10) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 0.0.0+sha.46d5afb
 - tool-esptoolpy @ 4.7.4
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- NimBLE-Arduino @ 1.4.2
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\cbuf.cpp.o
In file included from C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:8:
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.h:17:6: error: variable or field 'attachInterrupt' declared void
   17 | void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode);
      |      ^~~~~~~~~~~~~~~
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.h:17:22: error: 'uint8_t' was not declared in this scope
   17 | void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode);
      |                      ^~~~~~~
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.h:12:1: note: 'uint8_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   11 | #include <functional>
  +++ |+#include <cstdint>
   12 |
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.h:17:61: error: expected primary-expression before 'intRoutine'
   17 | void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode);
      |                                                             ^~~~~~~~~~
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.h:17:73: error: expected primary-expression before 'int'
   17 | void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode);
      |                                                                         ^~~
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-adc.c.o
*** [.pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o] Error 1
In file included from c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\xtensa-esp32-elf\no-rtti\bits\gthr.h:148,
                 from c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ext\atomicity.h:35,
                 from c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\bits\shared_ptr_base.h:61,
                 from c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\bits\shared_ptr.h:53,
                 from c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\memory:77,
                 from C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:25:
c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h: In function 'int __gthread_mutex_timedlock(__gthread_mutex_t*, const __gthread_time_t*)':
c:\users\UserX\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:769:12: error: 'pthread_mutex_timedlock' was not declared in this scope; did you mean 'pthread_mutex_trylock'?
  769 |     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);
      |            ^~~~~~~~
cc1.exe: warning: '-Werror=' argument '-Werror=reorder' is not valid for C
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp: In member function 'uint32_t EspClass::getFlashChipSize()':
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:256:17: warning: converting a packed 'esp_image_header_t' pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  256 |     if(flashRead(0x1000, (uint32_t*)&fhdr, sizeof(esp_image_header_t)) && fhdr.magic != ESP_IMAGE_HEADER_MAGIC) {
      |        ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/app_update/esp_ota_ops.h:23,
                 from C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:30:
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bootloader_support/esp_image_format.h:72:16: note: defined here
   72 | typedef struct {
      |                ^
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp: In member function 'uint32_t EspClass::getFlashChipSpeed()':
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:265:17: warning: converting a packed 'esp_image_header_t' pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  265 |     if(flashRead(0x1000, (uint32_t*)&fhdr, sizeof(esp_image_header_t)) && fhdr.magic != ESP_IMAGE_HEADER_MAGIC) {
      |        ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bootloader_support/esp_image_format.h:72:16: note: defined here
   72 | typedef struct {
      |                ^
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp: In member function 'FlashMode_t EspClass::getFlashChipMode()':
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/cores/esp32/Esp.cpp:274:17: warning: converting a packed 'esp_image_header_t' pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Waddress-of-packed-member]
  274 |     if(flashRead(0x1000, (uint32_t*)&fhdr, sizeof(esp_image_header_t)) && fhdr.magic != ESP_IMAGE_HEADER_MAGIC) {
      |        ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/UserX/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bootloader_support/esp_image_format.h:72:16: note: defined here
   72 | typedef struct {
      |                ^
*** [.pio\build\esp32dev\FrameworkArduino\Esp.cpp.o] Error 1
========================================================================================================== [FAILED] Took 10.48 seconds ==========================================================================================================
```

I tried other versions like 1.3.x but that also fails with other errors. Any idea what Im doing wrong?

Thanks, regards,
Bastiaan

h2zero, 2024-11-04:
Those errors all appear to be in the Arduino core, try deleting your .pio folder. Thats also a very old core version, I would recommend updating.

BasvanH, 2024-11-05:
Are you referring to the PlatformIO core, cause I'm running the latest version [6.1.16](https://docs.platformio.org/en/latest/core/history.html)

h2zero, 2024-11-05:
Sorry, I meant the arduino core version

BasvanH, 2024-11-05:
No worries, I appreciate the help.

Updated to framework-arduinoespressif32 @ 3.0.7, still wont build. Any idea?

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing h2zero/NimBLE-Arduino @ 1.4.2
Unpacking  [####################################]  100%
Library Manager: NimBLE-Arduino@1.4.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2024.7.10) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.7+sha.3bfa3e0
 - tool-esptoolpy @ 4.7.4
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
KeyError: 'framework-arduinoespressif32-libs':
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\UserX\.platformio\platforms\espressif32\builder\main.py", line 359:
    target_elf = env.BuildProgram()
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 684:
    return method(*args, **kw)
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\UserX\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 49:
    SConscript(join(FRAMEWORK_DIR, "tools", "platformio-build.py"))
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 684:
    return method(*args, **kw)
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\UserX\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\UserX\.platformio\packages\framework-arduinoespressif32\tools\platformio-build.py", line 38:
    FRAMEWORK_LIBS_DIR = platform.get_package_dir("framework-arduinoespressif32-libs")
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\UserX\.platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),
========================================================================================================== [FAILED] Took 12.38 seconds ==========================================================================================================

 *  The terminal process "C:\Users\UserX\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

[... 1205 lines cut ...]
.pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:457:4: error: 'esp_timer_handle_t' does not name a type; did you mean 'TimerHandle_t'?
  457 |    esp_timer_handle_t handle;
      |    ^~~~~~~~~~~~~~~~~~
      |    TimerHandle_t
[... cut ...]
*** [.pio\build\esp32dev\lib65f\NimBLE-Arduino\NimBLEDescriptor.cpp.o] Error 1
*** [.pio\build\esp32dev\lib65f\NimBLE-Arduino\NimBLEAdvertisedDevice.cpp.o] Error 1
*** [.pio\build\esp32dev\lib65f\NimBLE-Arduino\NimBLEAdvertising.cpp.o] Error 1
*** [.pio\build\esp32dev\lib65f\NimBLE-Arduino\NimBLEClient.cpp.o] Error 1
========================================================================================================== [FAILED] Took 20.72 seconds ==========================================================================================================
```

h2zero, 2024-11-05:
Unfortunately espressif dropped support for pio in 3.x. 
Try this:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = h2zero/NimBLE-Arduino@1.4.2
```

I just removed the platform_packages, it should compile fine.

BasvanH, 2024-11-10:
Well that pulled 3.0.5 as well, I changed it to:
```
[env:esp32dev]
platform = espressif32@6.9.0
board = esp32dev
framework = arduino
lib_deps = h2zero/NimBLE-Arduino@1.4.2
```
And now it pulls `framework-arduinoespressif32 @ 3.20017.0 (2.0.17)` and compiles without any errors.

closed 2024-11-19 by BasvanH