h2zero/NimBLE-Arduino #765: platformio registry weirdness with 1.4.3

gkoh, 2024-11-28:
I just tried to move from a earlier cut of master to 1.4.3 from the platformio registry.
For whatever reason, the tree that platformio provides is all kinds of weird.
Specifically, my compile fails due to missing virtual functions for `NimBLEServer::onIdentity()`

In `libdeps`, I can see the following in `CHANGELOG.md`:
```
## [1.4.3] 2024-11-27
```

however, `NimBLEServer.h` is completely missing `NimBLEServer::onIdentity()`.
Furthermore, `onAuthenicationComplete()` looks like:
```
virtual void onAuthenticationComplete(ble_gap_conn_desc* desc);
```
when it should be taking a `NimBLEConnInfo`.

My platformio looks like this:
```
lib_deps = 
  h2zero/NimBLE-Arduino@1.4.3
```

Crazily, if I do this instead:
```
lib_deps =
  https://github.com/h2zero/NimBLE-Arduino#1.4.3
```
I get exactly what is expected.

Something really weird is going on.

h2zero, 2024-11-28:
The platformio release is correct, the github release is not :( I chose the wrong branch

h2zero, 2024-11-28:
Fixed the github release now.

What you are expecting in your code is what is in the master, upcoming 2.0.0 release. 1.4.3 was just a minor maintenance.

gkoh, 2024-11-28:
> Fixed the github release now.
> 
> What you are expecting in your code is what is in the master, upcoming 2.0.0 release. 1.4.3 was just a minor maintenance.

Aha!
That explains a lot, including my confusion as to why a patch release made significant API changes.

gkoh, 2024-11-28:
OK, 1.4.3 tag now looks as expected, thank you!

closed 2024-11-28 by gkoh