h2zero/NimBLE-Arduino #775: NimBLEScanResults.getDevice(BLEAddress) not working in 2.0.0

AdyRock, 2024-12-02:
This code is working:
```
	BLEScan* pBLEScan = BLEDevice::getScan();

	const BLEAddress bleAddress( BLECommand->Address );
	Serial.printf( "Sending command to BLE device: %s\n", BLECommand->Address );
	uint64_t requestAddress = bleAddress;

	NimBLEScanResults results = pBLEScan->getResults();
	uint8_t numResults = results.getCount();
	const NimBLEAdvertisedDevice* pDevice = nullptr;
	for (int i = 0; i < numResults; i++)
	{
		pDevice = results.getDevice( i );

		uint64_t deviceAddress = pDevice->getAddress();
		if (deviceAddress == requestAddress)
		{
			break;
		}
	}

	if ( pDevice != nullptr )
```

This code doesn't (but it did prior to 2.0.0):
```
	BLEScan* pBLEScan = BLEDevice::getScan();

	const BLEAddress bleAddress( BLECommand->Address );
	Serial.printf( "Sending command to BLE device: %s\n", BLECommand->Address );

	NimBLEScanResults results = pBLEScan->getResults();

	// Get the device (might be null if not found)
	const NimBLEAdvertisedDevice* pDevice = results.getDevice( bleAddress );

```

h2zero, 2024-12-02:
@AdyRock I see your issue now. In 2.0.0 the `getResults()` function takes a scan time parameter now, so if you didn't do a scan already then the results would be empty so you should either scan before calling this or provide a time to scan for when calling it.
This replaced the old blocking call of `NimBLEScan::start` and is one of the many breaking changes in 2.0.0.

h2zero, 2024-12-02:
`NimBLEAddress` has an == operator so it is not comparing the addresses of the variables. What I suspect is happening is the address you are using to find the device in the results does not have the same address type. This was a change in 2.0.0 as well because as per BLE spec "the addresses are not the same if the type is different, even if the address values are equal". Most address types from non-esp devices are random static so the address type used should be `1`.

h2zero, 2024-12-03:
Please see #777

closed 2026-03-17 by h2zero