h2zero/NimBLE-Arduino #776: NimBLEAddress == operator not working

chrisumbongo, 2024-12-03:
I have the following code where I am looking for a specific device by address, This works with the ArduinoBLE, but not with NimBle

```
  void onResult(const NimBLEAdvertisedDevice* advertisedDevice) 
  {
    Serial.printf("Advertised Device Result: %s \n", advertisedDevice->toString().c_str());
    Serial.println(advertisedDevice->getAddress().toString().c_str());
    Serial.println(SmartBMV.toString().c_str());

    if(advertisedDevice->getAddress().toString() == SmartBMV.toString())
    {
      Serial.printf("Found BMV\n");
    }
    if(advertisedDevice->getAddress() == SmartBMV)
    {
      Serial.printf("Found BMV\n");
    }  
  }
```

when I use toString it works, when I use the NimBLEAddress it doesn't.

h2zero, 2024-12-03:
Hello @chrisumbongo I am assuming you have installed the accidentally released version 2 (which is pending removal)? 
You will need to account for the address type in version 2, so your `SmartBMV` needs to also have the address type set correctly, it is probably `1` (random).

Please see this post: https://github.com/h2zero/NimBLE-Arduino/issues/775#issuecomment-2512043520

chrisumbongo, 2024-12-03:
aah ok, that make sense, which version should I use?

chrisumbongo, 2024-12-03:
1.4.3 duh... it's on the main page, my bad, thanks!

closed 2024-12-03 by chrisumbongo

h2zero, 2024-12-03:
Just FYI, when 2.0.0 is properly released you will need to adjust your code. I just created #777 to help.

chrisumbongo, 2024-12-04:
I reverted back to 1.4.3 but then the IDE just upgraded to 2.0.0 again... so I've done this:-

```if(advertisedDevice->getAddress().toString() == SmartBMV.toString())```

it works now and should work when 2.0.0 is ready.

h2zero, 2024-12-04:
I hope they remove that soon... https://github.com/arduino/library-registry/issues/5494#issuecomment-2508861916

`if(advertisedDevice->getAddress().toString() == SmartBMV.toString())` this works for sure, not ideal but functional 😄 

Can you share how the address for `SmartBMV` is constructed? I suspect that you will have a compile error in version 2 as the constructor will require the type to be specified