h2zero/NimBLE-Arduino #777: Notice about BLE Address comparisons

h2zero, 2024-12-03:
I'm seeing a trend in the issues since the mistaken and soon to be removed release of version 2.0.0 on the Arduino library manager where the address comparisons that were working are now not, so I will explain why here.

BLE has what is referred to as address types, there are `public`, `random static`, `random resolvable private`, and `non-resolvable private` addresses. 

For the purpose of this post we only need to know that there are 2 types of concern when devices are advertising `public; type = 0` and `random; type = 1`. 

### The BLE specification says:
> Whenever two device addresses are compared, the comparison shall include the device address type (i.e. if the two addresses have different types, they are different even if the two 48-bit addresses are the same).

### So, what changed? 
Prior to version 2, the address type was not compared and so a device with a random (type 1) address `11:22:33:44:55:66` would return equal to a device with a public (type 0) address `11:22:33:44:55:66` which as stated in the specification would be incorrect. 

### How to fix it?
If you are scanning for a specific device by address you will need to set the type when you create the address, this is done by providing the type parameter to the constructor: `NimBLEAddress myAddress("11:22:33:44:55:66", **TYPE; 0 or 1**);

### How to find the address type?
When scanning you can get the address type from the advertised device through either `NimBLEAdvertisedDevice::getAddressType()` or from the address itself `NimBLEAddress::getType()`.

Hope this helps, post any questions below.

thorrak, 2024-12-16:
Aren't the address types indicated in the address itself, based on the first octet?

h2zero, 2024-12-17:
The last 2 bits in the MSB only indicate the type of random address, public addresses can be anything.

thorrak, 2024-12-17:
Doesn’t the presence/absence of the bits indicate what the address type is though (1 vs 0)?  I guess what I’m wondering is how setting a type on the address would actually be communicated outside of the communication of the address itself.

h2zero, 2024-12-17:
Okay, now I have to get deep lol. 

In the underlying advertisement there are certain flags (not controllable from the app or the host), one of them is the address type, which is what scanners receive in the advertisement information area, that is over and above the data itself. 

The importance of this flag goes into how Bluetooth SIG operates. To have a public address you must pay $$$$$ to Bluetooth SIG, which Espressif has done, to secure a public address, which gives them ownership of the first 3 MSB's and the last 3 are distributable per device.  This gives them the ability to use a public address, which could have anything the Bluetooth SIG assigns them for the 3 MSB's, the bits do not matter at all for public addresses.

Consequently this is why you will see that the majority of devices have a random address type (1), it's free.