h2zero/NimBLE-Arduino #778: Using ``NimBLERemoteCharacteristic::getValue()`` in callbacks for ``NimBLERemoteCharacteristic::subscribe()``

Laxilef, 2024-12-04:
Hi,

I looked at PR #49 but couldn't figure out if I can use method ``NimBLERemoteCharacteristic::getValue()`` in callbacks?
The PR says that we should pass a ``timestamp`` to prevent re-reading, but is this relevant now?

Or can we just use such an expression now?
```C
          pChar->subscribe(true, [](NimBLERemoteCharacteristic* pChar, uint8_t* pData, size_t length, bool isNotify) {
            // ...
            auto value = pChar->getValue<short>();
            // ...
          });
```

h2zero, 2024-12-04:
@Laxilef you can freely call this function, it just returns the internally stored value, does not trigger a read.

Laxilef, 2024-12-04:
Awesome! Thanks

closed 2024-12-04 by Laxilef