h2zero/NimBLE-Arduino #779: Issue with Client-Server Connection and Pairing Using NimBLE Library

nitin-jethava, 2024-12-04:
I'm transitioning my BLE implementation from the BLEDevice library to the NimBLE library to optimize flash and RAM usage. Here's the issue I'm facing:  

#### Previous Library (BLEDevice):
- Implemented both server and client functionality.
- Used `client->connect(remote_address);` to connect to a remote device based on its Bluetooth address (`remote_address`).
- Configured pairing and bonding as follows:  
  ```cpp
  BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
  BLEDevice::setSecurityCallbacks(new _Pairing());

  pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
  pSecurity->setCapability(ESP_IO_CAP_IO);
  pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
  ```
- Pairing and bonding worked, and the client could securely connect to the server.

#### NimBLE Implementation:
- Created both server and client functionality with similar logic.
- For the remote device address, I tried both `desc->peer_ota_addr` and `desc->peer_id_addr` as the client address in `client->connect(remote_address);`.

**Current Issues:**
1. No pairing prompt appears when I connect the server using a mobile phone.
2. The client fails to connect to the server using the provided remote device addresses.

Looking for guidance on how to resolve these issues in NimBLE, especially for establishing pairing and secure client-server connections.

wuhuqif176, 2024-12-08:
Hello, I encountered a similar issue. When pairing my service device using an Android/win device, a “Pairing Request” window pops up. However, when using the NimBLE library to connect, there is no “Pairing Request” window that pops up, even though I have enabled secure pairing.

// Connect to the specified server address and set up automatic reconnection
pClient->connect(NimBLEAddress(serverAddress), true);

NimBLEDevice::setSecurityAuth(true, true, true);
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_DISPLAY);
pClient->setConnectTimeout(30);
if (pClient->secureConnection())
{
  Serial.println("Pairing successful"); // Pairing request success message
}
else
{
  Serial.println("Pairing failed");
  uint32_t err = pClient->getLastError();
  Serial.print("Error code: ");
  Serial.println(err, HEX); // Print error code
}
The serial terminal outputs “Pairing failed” with an error code ‘D’. The service device also does not show a “Pairing Request” window.

😭

h2zero, 2024-12-08:
Please test with the example here: https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/examples/NimBLE_Secure_Server/NimBLE_Secure_Server.ino and let me know if that works.

h2zero, 2024-12-08:
@nitin-jethava 
>The client fails to connect to the server using the provided remote device addresses.

If you are hard coding an address you must make sure the address type is correct, default is 0, so you probably need to set it to 1 (random).