h2zero/NimBLE-Arduino #780: Subscription to Notification stops working 2.0.0

danbITC, 2024-12-04:
The notifications will work initially, but it seems they timeout or just stop functioning at some point. I think it happens after 4-5 minutes.

I am running this code on an ESP32-S3

I setup the notifications like this (taken from an example on this repo)
```
if (pRead->canNotify()) {
        if (!pRead->subscribe(true, notifyReadCB)) {
            pClient->disconnect();
            LV_LOG_USER("Failed to subscribe to read characteristic");
        }
    }
```
The notifyReadCB looks like this. Modified from an example on this repo.
```void notifyReadCB(NimBLERemoteCharacteristic *pRemoteCharacteristic, uint8_t *pData, size_t length, bool isNotify) {
    // LV_LOG_USER("Notify/Indicate callback called");
    // LV_LOG_USER("isNotify: %s", isNotify ? "true" : "false");
    std::string str = (isNotify == true) ? "Notification" : "Indication";
    str += " from ";
    /** NimBLEAddress and NimBLEUUID have std::string operators */
    str += std::string(pRemoteCharacteristic->getClient()->getPeerAddress());
    // str += ": Service = " + std::string(pRemoteCharacteristic->getRemoteService()->getUUID());
    // str += ", Characteristic = " + std::string(pRemoteCharacteristic->getUUID());
    std::string dataStr;
    for (size_t i = 0; i < length; ++i) {
        char buf[4];
        snprintf(buf, sizeof(buf), "%02x", pData[i]);
        dataStr += buf;
        if (i < length - 1) {
            dataStr += " ";
        }
    }
    str += ", Value = " + dataStr;
    LV_LOG_USER(str.c_str());
    receiveBluetoothMessage(pData, length);
    // LV_LOG_USER("Connection Interval: %d    Connection Latency: %d    Connection Timeout: %d", conn.getConnInterval(), conn.getConnLatency(), conn.getConnTimeout());
}

The connection happens successfully, the connection parameter negotiation happens successfully. I don't see a disconnect event.
I know that the device that I am connecting to works normally with a phone app, so I don't believe that its something wrong with the device itself. It's a nordic softdevice (S132) which seems fairly mature.

I'm not quite sure where to start looking to resolve this issue as I am not intimately familiar with BLE protocols. If I could be pointed in the right direction I may be able to help resolve this with a PR. Do I need to have some sort of a heartbeat message to keep the connection alive?

h2zero, 2024-12-04:
2.0.0 was an accidental release and is still pending removal from the library manager unfortunately. I suggest reverting back to 1.4.3 until 2.0.0 is officially released with the documentation to help with this issue (many things have been changed).

danbITC, 2024-12-04:
Understood, I am intentionally developing with 2.0.0 to better understand all of the changes needed to make the upgrade and not using the library manager.

danbITC, 2024-12-09:
I do not, but I was able to band-aid the issue for now by sending a message every two minutes to trigger the notification.