h2zero/NimBLE-Arduino #781: Does scanning continue while advertising

I-Connect, 2024-12-04:
Hi,

We have several devices that communicate via ble advertisements and I am struggling a bit with advertisements sometimes not being "seen" by a another device and am trying to understand what the different settings do and how the ble stack exactly works.

I see in the examples:
```
// Start advertising
pAdvertising->start();
Serial.println("Advertizing started for 10s ...");
delay(10000);
pAdvertising->stop();

```
So it is advertising for 10 seconds. Suppose I have enabled scanning indefinitely before the above code :
`NimBLEScan::start(0, ....)`

Does the 10 second advertisement interrupt the scanning completely for 10 seconds or is the ble stack that intelligent that it continues scanning in between the advertisements defined by the advertisement interval?

And a second question, what would be the recommended values for these methods knowing a device is both scanning and advertising (power usage is not an issue)
[image]

h2zero, 2024-12-04:
Both can be done at the same time but you will want to adjust your window and interval for both scan and advertise so that there is enough time slicing available for each.

You'll need to experiment with the values but I'd start with a 50/50 mix, params :interval 100ms window 50ms for scan and advertising both as a starting point.

I-Connect, 2024-12-04:
thx for the quick answer :-)

What I forgot to mention is that the data is not only sent periodically. 
I advertise every 60 secs all data, but want to advertise immediately when a switch is flipped on a device (and want to reduce the chance of missing this to a minimum).

If I use your suggestion there is a chance a 50ms advertising does not get detected in a 50/50 scan/advertise ...?

h2zero, 2024-12-04:
That's not the advertising duration, just the time inside the interval. The number of intervals is the duration / interval time. So your 10 second duration / 100ms interval time = 100 intervals, so the advertising is happening for 50ms 100 times over 10 seconds.

I-Connect, 2024-12-04:
I am sorry, I have a lot of trouble getting my head around this :-)

This would generate a lot of advertisements with the same data (I only need 1 to go through to the other device)
I am not sure what the bandwidth is of using BLE like this...
Wouldn't  so many advertisements also have influence on receiving signals from other sensors that are also advertising data?

so would it look like this in code?:

//Initialisation
```
NimBLEScan::start(0, ....); // indefinite scanning (if I understand correctly it stops and starts automatically when advertising?)
NimBLEAdvertising::setMinInterval(100);
NimBLEAdvertising::setMaxInterval(100);
```

And then in the loop() when it needs to advertise:
```
if(needsToAdvertise){
  pAdvertising->stop();  //stop any running advertisement as maybe data has changed
  pAdvertising->setAdvertisementData(advertisementData);
  pAdvertising->start(10);
}
```

Or maybe I could make a setup that the receiving device echo's the data and when the sending device receives it and it is the same it stops advertising...?

h2zero, 2024-12-04:
Sorry, I was mixing terms with scanning.

The min and max advertising interval is was I meant to refer to, which should be different values. The difference between them gives the rest of the ble stack time to work with the radio. 

```
NimBLEScan::start(0, ....); // indefinite scanning (if I understand correctly it stops and starts automatically when advertising?)
NimBLEScan::setInterval(100); 
NimBLEScan::setWindow(50); // scan only for half the interval time to allow for advertising to take place.
NimBLEAdvertising::setMinInterval(50); 
NimBLEAdvertising::setMaxInterval(100); // time difference for scan etc..
```

h2zero, 2025-04-23:
Closing this as it was not an issue with the library.