h2zero/NimBLE-Arduino #783: nRF52840 extended advertising in Arduino IDE fails compile with undefined reference to `ble_gap_ext_adv_configure'

stuartlangridge, 2024-12-07:
I'm trying to use an nRF52840 dongle from the Arduino IDE with NimBLE-Arduino. I've successfully compiled, uploaded, and run a simple example, which advertised a BLE service, so that's good. I'm now trying to use extended advertising. I've set `#define CONFIG_BT_NIMBLE_EXT_ADV 1` in nimconfig.h as requested, but when trying to compile the built-in `NimBLE_extended_server` example (and the compile passes the ifdef check for CONFIG_BT_NIMBLE_EXT_ADV, so I definitely changed the correct nimconfig.h), the build dies with the following error:

/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 4 bytes
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 4 bytes
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../…ges/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/84F70A87C9A9937FD2F9FD4F2A07A716/libraries/NimBLE-Arduino/NimBLEExtAdvertising.cpp.o: in function `NimBLEExtAdvertising::stop()':
/home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:254: undefined reference to `ble_gap_ext_adv_clear'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

This looks like it's missing some library definitions of some sort? Are there other config options that I need to turn on, or do I need to actually rebuild the library somehow after editing nimconfig.h? I'm a bit baffled by this.

h2zero, 2024-12-07:
Is this on the release branch or master?

stuartlangridge, 2024-12-07:
I think the release branch? I added NimBLE-Arduino from the Arduino IDE library manager, which claims that "2.0.0" is installed.

h2zero, 2024-12-07:
That release is pending removal, I would suggest that you revert to 1.4.3 and try again. 2.0.0 is not ready yet and was accidently released by selecting the wrong branch when 1.4.3 was initially released.

stuartlangridge, 2024-12-07:
Unfortunately, that hasn't helped; I have 1.4.3 installed now and I get the same error when compiling `NimBLE_extended_server.ino`. (I've re-edited nimconfig.h to enable extended advertising.)

```
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/2DF17F8CE4D2F55AE5A44B20DF2F831D/libraries/NimBLE-Arduino/NimBLEExtAdvertising.cpp.o: in function `NimBLEExtAdvertising::start(unsigned char, int, int)':
/home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:163: undefined reference to `ble_gap_ext_adv_start'
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/2DF17F8CE4D2F55AE5A44B20DF2F831D/libraries/NimBLE-Arduino/NimBLEExtAdvertising.cpp.o: in function `NimBLEExtAdvertising::stop()':
/home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:258: undefined reference to `ble_gap_ext_adv_clear'
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/2DF17F8CE4D2F55AE5A44B20DF2F831D/libraries/NimBLE-Arduino/NimBLEExtAdvertising.cpp.o: in function `NimBLEExtAdvertising::setInstanceData(unsigned char, NimBLEExtAdvertisement&)':
/home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:65: undefined reference to `ble_gap_ext_adv_configure'
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:95: undefined reference to `ble_gap_ext_adv_rsp_set_data'
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:108: undefined reference to `ble_gap_ext_adv_set_addr'
/home/aquarius/.arduino15/packages/h2zero/tools/gcc-arm-none-eabi/9.3.1-1/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/aquarius/Programs/Mine/digimini3/libraries/NimBLE-Arduino/src/NimBLEExtAdvertising.cpp:97: undefined reference to `ble_gap_ext_adv_set_data'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

h2zero, 2024-12-07:
I just built the example with 1.4.3 and 2.0.0, please give this sketch a try:
[sketch_dec07c.zip](https://github.com/user-attachments/files/18048939/sketch_dec07c.zip)

stuartlangridge, 2024-12-07:
ok! That compiled fine. So that suggests that I need that build_opt.h file in my sketch, and indeed adding it to my own sketch makes that compile as well. Thank you!
Apologies if it is obvious to most people that they need to define build opts in this way; I didn't know this.

h2zero, 2024-12-07:
Usually it would work by editing the `nimconfig.h` file but `build_opts.h` is the recommended way.

h2zero, 2024-12-07:
@stuartlangridge can we close this?

stuartlangridge, 2024-12-07:
Yes, I'll close it. I'm having other problems, but they're not related to this bug so this can be closed!

closed 2024-12-07 by stuartlangridge