h2zero/NimBLE-Arduino #784: NOTICE: Release 2.0.0 was accidental

h2zero, 2024-12-07:
Sorry if this caused any trouble, when releasing 1.4.3 I forgot to change the branch to the release branch and accidentally released from the master branch. This caused the Arduino library manager to pick it up overnight before I could delete the release. 

So please consider that a beta version if you would like to use it, a new release with the documentation will be coming soon to replace it.

h2zero, 2024-12-15:
Release 2.1.0 is out, this can now be closed. Please update your application to use 2.1.0 and read the [migration guide}(https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/1.x_to2.x_migration_guide.md) to update your code.

h2zero, 2024-12-16:
2.1.0 is released now, sorry for the confusion, please update and read the migration guide.

davidlao2k, 2024-12-17:
NimBLEClient setConnectTimeout is a breaking change. It's now millisecond rather than second. It would be good to add that to the migration doc

h2zero, 2024-12-17:
@davidlao2k Sorry, the change log did state:
> All functions that take a time input parameter now expect the value to be in milliseconds instead of seconds.
 I will try to be more specific in the future but I rushed this a bit due to the Arduino library manager issue.

closed 2025-01-04 by h2zero