h2zero/NimBLE-Arduino #785: ESP32-S3 using NimBLE 1.4.3 and ESP-IDF 3.0.3 encounters an error with 128-bit UUID

wuhuqif176, 2024-12-08:
I wrote a program following the example BLE_client in the Refactored_original_examples folder.
#include <NimBLEDevice.h>

const char *serverAddress = "d8:10:68:1e:7a:8c"; // Target server's MAC address

static BLEUUID serviceUUID("00018000-0000-1000-8000-00805f9b34fb");

NimBLERemoteService *pSvc = nullptr;
NimBLERemoteCharacteristic *pChr = nullptr;
NimBLERemoteDescriptor *pDsc = nullptr;

int scanTime = 5;  // Scan time in seconds
BLEScan *pBLEScan; // BLE scan object pointer

uint16_t CAMERA_MANUFACTURER_ID = 0x012D; // Camera manufacturer ID

// Custom advertised device callback class for handling scanned device information
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
  void onResult(BLEAdvertisedDevice *advertisedDevice) override
  {
    // Check if the device has manufacturer data and the data size is at least 4 bytes
    if (advertisedDevice->haveManufacturerData() && advertisedDevice->getManufacturerData().size() >= 4)
    {
      std::string manData = advertisedDevice->getManufacturerData(); // Get manufacturer data
      // Check if the manufacturer data matches a specific value
      if (manData[0] == 0x2D && manData[1] == 0x01 && manData[2] == 0x03 && manData[3] == 0x00)
      {
        // Get and print the device name
        std::string deviceName = advertisedDevice->getName();
        Serial.print("Device Name: ");
        Serial.println(deviceName.c_str());

        // Get and print the device address
        BLEAddress macAddress = advertisedDevice->getAddress();
        Serial.print("Device Address: ");
        Serial.print(macAddress.toString().c_str());
        Serial.println("");
      }
    }
  }
};

void setup()
{
  Serial.begin(115200);                                      // Initialize serial port with a baud rate of 115200
  Serial.println("Scanning for devices with specific manufacturer data..."); // Print prompt message
  NimBLEDevice::init("ESP32");

  pBLEScan = BLEDevice::getScan();                                           // Create a new scan object
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()); // Set advertised device callbacks
  pBLEScan->setActiveScan(true);                                             // Set to active scan
  pBLEScan->setInterval(100);                                                // Set scan interval
  pBLEScan->setWindow(99);

  // Start scanning for scanTime seconds, do not repeat scan
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: "); // Print the number of devices found
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!"); // Scan completion prompt

  // Create NimBLEClient object for connecting to the server
  NimBLEClient *pClient = BLEDevice::createClient();

  // Connect to the specified server address and set auto-reconnect
  pClient->connect(NimBLEAddress(serverAddress), true);
}

void loop()
{
  // Get the connected client object based on the server address
  NimBLEClient *pClient = BLEDevice::getClientByPeerAddress(NimBLEAddress(serverAddress));
  if (pClient->isConnected()) // Check if the client object is not null and is connected
  {
    Serial.println("Still connected to the server."); // Connection status prompt

    BLERemoteService *pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr)
    {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
    }
    else
    {
      Serial.println("Not connected to the server."); // Not connected status prompt
      uint32_t err = pClient->getLastError();
      Serial.print("Error Code: ");
      Serial.println(err, HEX); // Print error code
    }
    delay(5000); // Wait for 5 seconds
  }
}
If I do this, the program crashes when it executes the line:
BLERemoteService *pRemoteService = pClient->getService(serviceUUID);
with the error:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42010627  PS      : 0x00060030  A0      : 0x8200a2b9  A1      : 0x3fcebf00
A2      : 0x00000000  A3      : 0x3fc9b854  A4      : 0x00060020  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x820099ca  A9      : 0x3fcebef0  
A10     : 0x3fc9b848  A11     : 0x00000000  A12     : 0x00000038  A13     : 0x3fc9b880  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42010624:0x3fcebf00 0x4200a2b6:0x3fcebf20 0x42003262:0x3fcebf70 0x420032b5:0x3fcebfa0 0x42002479:0x3fcebfe0 0x42016658:0x3fcec050 0x40382516:0x3fcec070

ELF file SHA256: 16d67687dc5891d0

If I change the 128-bit UUID to a 16-bit UUID, the program runs normally:

```cpp
BLERemoteService *pRemoteService = pClient->getService("1800");

I am a beginner. Is there a problem with my code? Thank you.

This is my environment

[env:esp32-s3-devkitm-1]
platform = espressif32
board = esp32-s3-devkitm-1
framework = arduino
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
	framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.3/esp32-arduino-libs-3.0.3.zip

lib_deps = 
	h2zero/NimBLE-Arduino@^1.4.3

h2zero, 2024-12-08:
Looks like the crash probably occurs after that line when you try to access it's functions. You should check if  `pRemoteService == nullptr` before accessing it. The uuid was not found on the server so it's not returning a valid pointer. This library will automatically try with 16/128bit uuid depending on what failed but this didn't work is because your 128bit uuid is not correct, you have `00018000-0000-1000-8000-00805f9b34fb`, try changing it to ` 00001800-0000-1000-8000-00805f9b34fb`.

wuhuqif176, 2024-12-09:
“Thanks, this is my issue, my OCR has deceived me, and I did not carefully check it.”

closed 2024-12-09 by wuhuqif176