h2zero/NimBLE-Arduino #786: ESP32-S3 pairing using Just Works method failed

wuhuqif176, 2024-12-08:
Hello, I’m having a bit of trouble pairing my camera with the esp32-s3. I’m trying to initiate pairing using the Just Works method, but it seems my camera is not receiving the request.

Here is the key part of the code:

```
NimBLEDevice::setSecurityAuth(true, false, true);
pClient->connect(NimBLEAddress(serverAddress), true);
pClient->secureConnection()
uint32_t err = pClient->getLastError();
Serial.println(err, HEX);  
```

The serial terminal returns ‘D’.

But I can connect and respond normally using Android/Windows devices.
[image]

[image]

Here is my complete code:

```

#include <NimBLEDevice.h>

const char *serverAddress = "d8:10:68:1e:7a:8c"; // Target server's MAC address

int scanTime = 5;  // Scan time in seconds
BLEScan *pBLEScan; // BLE scan object pointer

uint16_t CAMERA_MANUFACTURER_ID = 0x012D; // Camera manufacturer ID

// Custom advertised device callback class for handling scanned device information
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
  void onResult(BLEAdvertisedDevice *advertisedDevice) override
  {
    // Check if the device has manufacturer data and the data size is at least 4 bytes
    if (advertisedDevice->haveManufacturerData() && advertisedDevice->getManufacturerData().size() >= 4)
    {
      std::string manData = advertisedDevice->getManufacturerData(); // Get manufacturer data
      // Check if the manufacturer data matches a specific value
      if (manData[0] == 0x2D && manData[1] == 0x01 && manData[2] == 0x03 && manData[3] == 0x00)
      {
        // Get and print the device name
        std::string deviceName = advertisedDevice->getName();
        Serial.print("Device Name: ");
        Serial.println(deviceName.c_str());

        // Get and print the device address
        BLEAddress macAddress = advertisedDevice->getAddress();
        Serial.print("Device Address: ");
        Serial.print(macAddress.toString().c_str());
        Serial.println("");
      }
    }
  }
};

void setup()
{
  Serial.begin(115200); // Initialize serial communication, set baud rate to 115200
  NimBLEDevice::init("ESP32");
  pBLEScan = BLEDevice::getScan();                                           // Create a new scan object
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()); // Set advertised device callbacks
  pBLEScan->setActiveScan(true);                                             // Set to active scan
  pBLEScan->setInterval(100);                                                // Set scan interval
  pBLEScan->setWindow(99);
  // Start scanning, set scan time to scanTime seconds, do not repeat scanning
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: "); // Print the number of devices found
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!"); // Scan completion message

  // Create a NimBLEClient object for connecting to the server
  NimBLEClient *pClient = BLEDevice::createClient();

  NimBLEDevice::setSecurityAuth(true, false, true);
  // Connect to the specified server address and set auto-reconnect
  pClient->connect(NimBLEAddress(serverAddress), true);

  Serial.println("Starting pairing process");

  pClient->setConnectTimeout(30);
  if (pClient->secureConnection())
  {
    Serial.println("Pairing successful"); // Pairing request successful message
  }
  else
  {
    Serial.println("Pairing failed");
    uint32_t err = pClient->getLastError();
    Serial.print("Error Code: ");
    Serial.println(err, HEX); // Print error code
  }
}

void loop()
{
}

```

h2zero, 2024-12-08:
Does the camera support "just-works" pairing? The error code is suggesting a timeout so it looks like it's not responding.

wuhuqif176, 2024-12-09:
> 相机是否支持“just-works”配对？错误代码建议超时，因此看起来没有响应。

I can basically confirm that my device uses the just-works pairing method, and I’ve tried several other pairing methods without any response from my device. However, pairing with Android/Windows devices works normally. Moreover, I found another person online who wrote a program using the adafruit_feather_nrf52840 development board. In his code, I saw that he pairs the device using:
```
if (_attempt_pairing == true)
{
    conn->requestPairing();
}
```
without adding any parameters. But when I tried using the 
```
pClient->secureConnection() 
```
function from the nimble library, it did not initiate a pairing request.

wuhuqif176, 2024-12-09:
I couldn’t find a function in Nimble to query the pairing method of the connected device. However, it seems that Nimble handles secure pairing automatically, and I’m not sure why it’s not working with my device.

h2zero, 2024-12-09:
Try changing this: `NimBLEDevice::setSecurityAuth(true, false, true);` to `NimBLEDevice::setSecurityAuth(true, false, false);`

wuhuqif176, 2024-12-21:
Thank you for your suggestion, but I still couldn’t pair after trying. During this period, I tried many methods. I found that pairing is possible when using the Bluedroid library with the BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT) method. However, when switching to the NimBLE library, it still does not work.
Here is my code:
Bluedroid ：
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5;        // Scan time in seconds
int actionGapTime = 250; // Action interval time in milliseconds, a typical debounce time

// To-do: Add more camera models
int cameraModelsSize = 1;
String cameraModels[] = {
    "ZV-E10",
};

static BLEScan *pBLEScan;
static BLERemoteCharacteristic *pRemoteCharacteristic;
static BLEClient *pClient;
static BLEAddress *pServerAddress;

bool doConnect = false; // Flag to indicate whether to connect
bool paired = false;    // Flag to indicate whether pairing has occurred
bool connected = false; // Flag to indicate whether the connection is established

// Callback class for advertised devices
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {
        // Get the device name
        String name = advertisedDevice.getName().c_str();

        // Check if the device is in the cameraModels list
        for (int i = 0; i < cameraModelsSize; i++)
        {
            if (name.indexOf(cameraModels[i]) != -1)
            {
                pServerAddress = new BLEAddress(advertisedDevice.getAddress());

                // Stop scanning
                pBLEScan->stop();
                Serial.println("Device found!");

                doConnect = true;
            }
        }
    }
};

// Security callback class
class MySecurity : public BLESecurityCallbacks
{
    bool onConfirmPIN(uint32_t pin)
    {
        Serial.println("Confirm PIN");
        return true;
    }

    uint32_t onPassKeyRequest()
    {
        Serial.println("Passkey request");
    }

    void onPassKeyNotify(uint32_t pass_key)
    {
        Serial.println("Passkey notify");
    }
[... 265 lines cut ...]
{
  Serial.begin(115200); // Initialize serial communication, set baud rate to 115200
  Serial.println("Scanning for devices with specific manufacturer data..."); // Print prompt message
  NimBLEDevice::init("");
  NimBLEDevice::setSecurityAuth(false, false, true); // Set encryption level
  NimBLEDevice::setPower(ESP_PWR_LVL_P9);            // Set power level to +9db
  NimBLEScan *pScan = NimBLEDevice::getScan();
  pScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()); // Set advertised device callbacks
  pScan->setActiveScan(true);
  pScan->setInterval(45);
  pScan->setWindow(15);
  pScan->start(scanTime, scanEndedCB);
}

void loop()
{
  if (doConnect == true)
  {
    doConnect = false;
    // Found a device we want to connect to, connect now
    connectToServer();
  }
}

```

h2zero, 2024-12-21:
Okay lets try some changes here, since the issue is a timeout occurring we should look at the connection parameters.
Lets start with:
```
  bool onConnParamsUpdateRequest(NimBLEClient *pClient, const ble_gap_upd_params *params)
  {
    if (params->itvl_min < 24)
    { /** 1.25ms units */
      return false;
    }
    else if (params->itvl_max > 40)
    { /** 1.25ms units */
      return false;
    }
    else if (params->latency > 2)
    { /** Number of intervals allowed to skip */
      return false;
    }
    else if (params->supervision_timeout > 100)
    { /** 10ms units */
      return false;
    }

    return true;
  };
```
Change this to just return `true` so the camera can have the parameters it likes:
```
  bool onConnParamsUpdateRequest(NimBLEClient *pClient, const ble_gap_upd_params *params)
  {
    return true;
  };
```
Now lets remove these lines:
```
    pClient->setConnectionParams(12, 12, 0, 51);
    pClient->setConnectTimeout(5);
```
After this please set the log level to debug and see what happens.

wuhuqif176, 2024-12-22:
After attempting to delete, I still cannot connect. Here is my log:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1654
load:0x403b6000,len:0xed4
load:0x403ba000,len:0x32dc
entry 0x403b632c
I (25) boot: ESP-IDF v4.4.1-405-g6c5fb29c2c 2nd stage bootloader
I (25) boot: compile time 04:22:20
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (79) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (86) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (94) boot:  4 uf2              factory app      00 00 00410000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (109) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1934ch (103244) map
I (629) esp_image: segment 1: paddr=00029374 vaddr=3fc96e40 size=04a94h ( 19092) load
I (633) esp_image: segment 2: paddr=0002de10 vaddr=40374000 size=02208h (  8712) load
I (636) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=5ae1ch (372252) map
I (699) esp_image: segment 4: paddr=0008ae44 vaddr=40376208 size=10c2ch ( 68652) load
I (720) boot: Loaded app from partition at offset 0x10000
I (721) boot: Disabling RNG early entropy source...
[   739][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface CDC enabled
E (747) esp_core_dump_flash: No core dump partition found!
E (747) esp_core_dump_flash: No core dump partition found!
[   748][D][esp32-hal-tinyusb.c:569] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEDevice: Setting bonding: 0, mitm: 0, sc: 1
D NimBLEDevice: >> setPower: 11 (type: 11)
D NimBLEDevice: << setPower
D NimBLEScan: >> start: duration=5
D NimBLEScan: << start()
I NimBLEScan: New advertiser: d8:10:68:1e:7a:8c
I NimBLEScan: Updated advertiser: d8:10:68:1e:7a:8c
D NimBLEScan: >> stop()
D NimBLEClient: >> connect(d8:10:68:1e:7a:8c)
D NimBLEScan: << stop()
D NimBLEClient: Got Client event 
I NimBLEClient: Connected event
D NimBLEClient: Got Client event 
I NimBLEClient: mtu update event; conn_handle=1 mtu=255
I NimBLEClient: Connection established
D NimBLEClient: >> deleteServices
D NimBLEClient: << deleteServices
D NimBLEClient: << connect()
D NimBLEClient: >> getRssi()
D NimBLEClient: >> getService: uuid: 8000ff00-ff00-ffff-ffff-ffffffffffff
D NimBLEClient: >> retrieveServices
D NimBLEClient: Service Discovered >> status: 0 handle: 66
D NimBLERemoteService: >> NimBLERemoteService()
D NimBLERemoteService: << NimBLERemoteService(): 8000ff00-ff00-ffff-ffff-ffffffffffff
D NimBLEClient: Service Discovered >> status: 14 handle: -1
D NimBLEClient: << Service Discovered
D NimBLEClient: << retrieveServices
D NimBLERemoteService: >> getCharacteristic: uuid: 0000ff01-0000-1000-8000-00805f9b34fb
D NimBLERemoteService: >> retrieveCharacteristics() for service: 8000ff00-ff00-ffff-ffff-ffffffffffff
D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovered
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLERemoteService: >> retrieveCharacteristics() for service: 8000ff00-ff00-ffff-ffff-ffffffffffff
D NimBLEClient: Got Client event 
I NimBLEClient: Connection parameters updated.
D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 71
D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0xff01
D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
D NimBLERemoteService: << Characteristic Discovered
D NimBLERemoteService: << retrieveCharacteristics()
D NimBLEClient: >> secureConnection()
D NimBLEClient: Got Client event 
E NimBLEClient: secureConnection: failed rc=13
D NimBLEClient: Got Client event 
I NimBLEClient: disconnect; reason=531,
```

h2zero, 2024-12-22:
Thanks, one more thing to try as I suspect the error code here is actually from the peer as it is trying to secure the connection at the same time that you are then that code would mean the procedure is already in progress.
So let's remove this:
```
  if (pClient->secureConnection())
  {
    Serial.println("Pairing successful");
  }
  else
  {
    Serial.println("Pairing failed");
  }
```
And see what happens. Notice that you didn't need to force the connection security with Bluedroid.

wuhuqif176, 2024-12-29:
Translation: I want to confirm whether writing the hexadecimal value 0x0107 is done by using 
```uint8_t data[] = { 0x01, 0x07 }; pChr->writeValue(data, sizeof(data)).```

Then I encountered some issues. After deleting that line of code, the ESP32 still did not request pairing. After I requested a write, the serial terminal returned normally with 
```NimBLERemoteCharacteristic: >> writeValue(), length: 2,```
but the camera did not perform any action. It is supposed to execute a focusing operation, which I am quite certain about, as I have confirmed it through various methods.

If I use ```pChr->writeValue(data, sizeof(data), true),``` then the library will automatically execute 
```NimBLEClient: >> secureConnection()```, but the camera remains the same as before and does not receive any requests.

Another issue is that notifications cannot be received. However, after pairing using the Bluedroid library, if the pairing information is not reset on the camera, then the return messages for the subscribed characteristics can be received.

h2zero, 2025-01-02:
You may need to reverse the byte order, generally all data over BLE is sent in little endian format. Try writing the value with respsonse request enabled.

closed 2025-01-04 by wuhuqif176