h2zero/NimBLE-Arduino #792: Windows cant connect to it

Volanaro, 2024-12-10:
so im trying a few diffrent things. i have some working code, that connects to android fine, but when i try and connect to windows it says:
Try connection your device again.

`#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>
#include <NimBLEUtils.h>
#include <driver/adc.h>

// HID Descriptor for an Xbox Controller (simplified)
const uint8_t xboxHIDDescriptor[] = {
    0x05, 0x01,                    // Usage Page (Generic Desktop)
    0x09, 0x05,                    // Usage (Game Pad)
    0xA1, 0x01,                    // Collection (Application)
    0xA1, 0x00,                    //   Collection (Physical)
    0x05, 0x09,                    //     Usage Page (Buttons)
    0x19, 0x01,                    //     Usage Minimum (1)
    0x29, 0x10,                    //     Usage Maximum (16)
    0x15, 0x00,                    //     Logical Minimum (0)
    0x25, 0x01,                    //     Logical Maximum (1)
    0x95, 0x10,                    //     Report Count (16)
    0x75, 0x01,                    //     Report Size (1)
    0x81, 0x02,                    //     Input (Data, Variable, Absolute)
    0x05, 0x01,                    //     Usage Page (Generic Desktop)
    0x09, 0x30,                    //     Usage (X)
    0x09, 0x31,                    //     Usage (Y)
    0x09, 0x32,                    //     Usage (Z)
    0x09, 0x35,                    //     Usage (Rz)
    0x15, 0x81,                    //     Logical Minimum (-127)
    0x25, 0x7F,                    //     Logical Maximum (127)
    0x75, 0x08,                    //     Report Size (8)
    0x95, 0x04,                    //     Report Count (4)
    0x81, 0x02,                    //     Input (Data, Variable, Absolute)
    0xC0,                          //   End Collection
    0xC0                           // End Collection
};

NimBLEHIDDevice* hid;
NimBLECharacteristic* inputReport;

void setup() {
    Serial.begin(115200);
    Serial.println("Starting Xbox Controller Emulation...");
NimBLEDevice::setSecurityInitKey(3);
    NimBLEDevice::init("Xbox Controller");
    NimBLEServer* pServer = NimBLEDevice::createServer();

    hid = new NimBLEHIDDevice(pServer);
    inputReport = hid->inputReport(1); // Report ID 1
    hid->manufacturer()->setValue("Custom HID Device");

    // Set VID and PID to Xbox values
    hid->pnp(0x01, 0x045E, 0x028E, 0x0100); // VID: 0x045E, PID: 0x028E, Version: 1.0
    hid->hidInfo(0x00, 0x01);               // Device supports Boot protocol

    // Set custom HID descriptor
    hid->reportMap((uint8_t*)xboxHIDDescriptor, sizeof(xboxHIDDescriptor));
    hid->startServices();

    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);
    pAdvertising->addServiceUUID(hid->hidService()->getUUID());
    pAdvertising->start();

    Serial.println("Xbox Controller Emulation Ready.");
}

void loop() {
    static uint8_t reportData[6] = {0}; // Report for buttons and joysticks
    reportData[0] = 0x01;               // Simulate Button 1 pressed
    reportData[1] = 0x00;               // Simulate no other buttons pressed
    reportData[2] = 0x00;               // X-axis center
    reportData[3] = 0x00;               // Y-axis center
    reportData[4] = 0x00;               // Z-axis center
    reportData[5] = 0x00;               // Rz-axis center

    inputReport->setValue(reportData, sizeof(reportData));
    inputReport->notify(); // Send the report
    delay(100);            // Update every 100ms
}
`

h2zero, 2024-12-10:
I think HID devices need to enable bonding to connect with windows

Volanaro, 2024-12-21:
brilliant thanks, i have vibrators etc working on my touch mouse project, this was more of a test to see if i can make it report on windows as a diffrent gamepad more than anything, i did google and a fair amount of people are saying that its not really "public" yet, i,e we can rename it as like  "2buttons 1 axis" but not actully name it something diffrent yet???

Volanaro, 2024-12-21:
yes thanks

closed 2024-12-21 by Volanaro