h2zero/NimBLE-Arduino #804: Output Report Function Not Working Properly in Release 2.1.0

Sab1e-GitHub, 2024-12-15:
Hello @h2zero ,
I am a contributor to the ESP32-BLE-Gamepad repository. We use your repository as a dependency, and I must say that your project is excellent! However, when adapting to Release 2.1.0, all other features work fine, but the Output Report doesn't seem to work properly. Specifically, when the host computer sends data, the following issue occurs:
```
Output Report send failed
Error writing report. Code: 1168, Message: Element not found.
```
You can find the host application here: [HIDOutputDemo_CSharp](https://github.com/Sab1e-GitHub/HIDOutputDemo_CSharp)
On the other hand, with the NimBLE Arduino version 1.4.3, I can use all the features without issues. I believe I have carefully checked the code, and it seems the issue is not with our repository. 

Additionally, I have tried printing to check if the OutputReportChr pointer is correctly obtained, and it looks fine:
```
    if (enableOutputReport) {
    BleGamepadInstance->outputGamepad = BleGamepadInstance->hid->getOutputReport(BleGamepadInstance->configuration.getHidReportId());
    BleGamepadInstance->outputReceiver = new BleOutputReceiver(outputReportLength);
    if (BleGamepadInstance->outputGamepad == nullptr) {
        Serial.println("Failed to create Output Report characteristic!");
        return;
    }
    BleGamepadInstance->outputGamepad->setCallbacks(BleGamepadInstance->outputReceiver);

    Serial.println("Enabled Output Report!");
    }
```
[image]

Here is the code I've modified in our repository: [ESP32-BLE-Gamepad/tree/master](https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad/tree/master)
And here is the old version of the repository code: [ESP32-BLE-Gamepad/tree/NimBLE-v1.4.3](https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad/tree/NimBLE-v1.4.3)

I am not very familiar with the Bluetooth low-level architecture, so I would greatly appreciate your help. Thank you very much!

h2zero, 2024-12-15:
@Sab1e-GitHub, thank you for reporting this, I believe I have resolved this in #805, please test it and let me know.

Sab1e-GitHub, 2024-12-15:
> @Sab1e-GitHub, thank you for reporting this, I believe I have resolved this in #805, please test it and let me know.

Sure, thank you very much! I'll try it now.

Sab1e-GitHub, 2024-12-15:
Hi @h2zero ,
#805 successfully resolved my issue! The Output Report functionality is now working properly. Thank you so much for your help!

PR #805 (2024-12-15): Fix NimBLEHIDDevice output report returning incorrect characteristic.
The input and output report characteristics share the same UUID and the `getOutputReport` was returning the input report characteristic. This change will return the correct characteristic by finding it with the index parameter.

Fixes #804
closed 2024-12-15 by Sab1e-GitHub