h2zero/NimBLE-Arduino #813: BLE UART Tx and Rx not working

ednieuw, 2024-12-17:
I can not get this code working with on my Arduino Nano ESP32 with Arduino core 2.0.18 arduino 5 with NimBLE V2.1.1
It works fine with NimBLE  V1.4.3

I tried using NimBLEDevice instead of BLEDevice but later discovered this is changed automatically with a #define

The call to pAdvertising->setName(BLEbroadcastName); as suggested in #772 seems not necessary because I see the name broadcasted in my app and I can connect to it. The BLEDevice::init(Mem.BLEbroadcastName); seems enough.

The terminal window of my Serial terminal app opens after connection and commands can be sent from my phone do not arrive in the Nano ESP32. Strings sent from the Nano ESP32 are not received by my phone.  

With the source below a connection with a [BLE serial terminal app](https://ednieuw.home.xs4all.nl/BLESerial/IOSappMain.html) on a IOS or Android device can be made
Commands can be sent to the ESP32 with the BLE serial terminal app and/or the serial terminal of the Arduino IDE. The commands and results are displayed on both terminals.

An UART example in the examples would be fine in a next release.

```
#include <NimBLEDevice.h>     
//--------------------------------------------                                                //
// BLE   //#include <NimBLEDevice.h>
//--------------------------------------------
BLEServer *pServer      = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected    = false;
bool oldDeviceConnected = false;
std::string ReceivedMessageBLE;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"                         // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

//----------------------------------------
// Common
//----------------------------------------
 
#define   MAXTEXT 255
char      sptext[MAXTEXT]; 
bool UseBLELongString = false;
char SSID[30];                                                                                // 
char Password[40];                                                                            // 
char BLEbroadcastName[30];                                                                    // Name of the BLE beacon

void setup() 
{
Serial.begin(115200);                                                                         // Setup the serial port to 115200 baud //
 int32_t Tick = millis(); 
 while (!Serial)  
 {if ((millis() - Tick) >5000) break;}  Tekstprintln("Serial started");                       // Wait max 5 seconds until serial port is started   
 strcpy(SSID,"");                                                                             // Default SSID
 strcpy(Password,"");                                                                         // Default password
 strcpy(BLEbroadcastName,"Edsoft");
 StartBLEService();     Tekstprintln("BLE started");                                          // Start BLE service

}

void loop() 
{
 CheckDevices();
}
//--------------------------------------------                                                //
// COMMON Check connected input devices
//--------------------------------------------
void CheckDevices(void)
{
 CheckBLE();                                                                                  // Something with BLE to do?
 SerialCheck();                                                                               // Check serial port every second 
}

//--------------------------------------------                                                //
// COMMON check for serial input
//--------------------------------------------
void SerialCheck(void)
[... 146 lines cut ...]
             if (InputString.length() >4 && InputString.length() <40)
              {  
               InputString.substring(1).toCharArray(Password,InputString.length());
               sprintf(sptext,"Password set: %s\n Enter @ to reset ESP32 and connect to WIFI and NTP\n WIFI and NTP are turned ON", Password); 
              }
             else sprintf(sptext,"**** Length fault. Use between 5 and 40 characters ****");
             break;   
    case 'C':
    case 'c': 
   
             if (InputString.length() >4 && InputString.length() <30)
               {  
                InputString.substring(1).toCharArray(BLEbroadcastName,InputString.length());
                sprintf(sptext,"BLE broadcast name set: %s", BLEbroadcastName); 
              }
            else sprintf(sptext,"**** Length fault. Use between 4 and 30 characters ****");
            break;                                
    default: break;
    }
  }  
 Tekstprintln(sptext);                                   
 InputString = "";
}

```

h2zero, 2024-12-20:
> The call to pAdvertising->setName(BLEbroadcastName); as suggested in https://github.com/h2zero/NimBLE-Arduino/issues/772 seems not necessary because I see the name broadcasted in my app and I can connect to it.

When your phone clears its cache or you use another phone the name won't show. It only works now because phones save the information about each device in it's cache.

Your callback function signatures are out of date and need to up updated for 2.x, please see the [1.x to 2.x migration guide](https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/1.x_to2.x_migration_guide.md)
Or look in the header files, https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/1.x_to2.x_migration_guide.md

ednieuw, 2024-12-23:
In the end the changes were simple but for an old fashioned C-programmer. pfft.

In NimBLE version 2.x.x these functions needed extra parameters in this sketch:

```
 void onConnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo) override {deviceConnected = true;Serial.println("Connected" ); };
 void onDisconnect(NimBLEServer* pServer, NimBLEConnInfo& connInfo, int reason) override {deviceConnected = false;Serial.println("NOT Connected" );}
 void onWrite(NimBLECharacteristic *pCharacteristic, NimBLEConnInfo& connInfo) override  

 add pAdvertising->setName(BLEbroadcastName);   after pAdvertising->addServiceUUID(SERVICE_UUID); 

```
Below a working sketch 
```

/* 
 Author      : Ed Nieuwenhuys https://github.com/ednieuw , https://ednieuw.nl
 December 2024

Sketch for Bluetooth UART (universal asynchronous receiver / transmitter) communication with a BLE serial terminal.

A serial terminal can be installed on a PC (like Termite) or on a IOS or Android device
For IOS:  BLESerial nRF or BLEserial Pro   
For Android:  Serial Bluetooth Terminal

In this example entries in the serial terminal can be used to set an SSID, password and BLE broadcast name.

******
Compile with NimBLE version 2.x
******

------------------------------------------------------------------------------------------------
*/

#include <NimBLEDevice.h>     
//--------------------------------------------                                                //
// BLE   //#include <NimBLEDevice.h>
//--------------------------------------------
NimBLEServer *pServer      = NULL;
NimBLECharacteristic * pTxCharacteristic;
bool deviceConnected    = false;
bool oldDeviceConnected = false;
std::string ReceivedMessageBLE;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"                         // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

//----------------------------------------
// Common
//----------------------------------------
 
#define   MAXTEXT 255
char      sptext[MAXTEXT]; 
bool UseBLELongString = false;                                                                // Set to true for faster communication on IOS devices
char SSID[30];                                                                                // 
char Password[40];                                                                            // 
char BLEbroadcastName[30];                                                                    // Name of the BLE beacon

//----------------------------------------
// Setup
//----------------------------------------
void setup() 
{
Serial.begin(115200);                                                                         // Setup the serial port to 115200 baud //
 int32_t Tick = millis(); 
 while (!Serial)  
 {if ((millis() - Tick) >5000) break;}  Tekstprintln("Serial started");                       // Wait max 5 seconds until serial port is started   
 strcpy(SSID,"");                                                                             // Default SSID
 strcpy(Password,"");                                                                         // Default password
[... 171 lines cut ...]
//--------------------------------------------                                                //
// BLE Start BLE Service
//------------------------------
void StartBLEService(void)
{
 NimBLEDevice::init(BLEbroadcastName);                                                        // Create the BLE Device
 pServer = NimBLEDevice::createServer();                                                      // Create the BLE Server
 pServer->setCallbacks(new MyServerCallbacks());
 BLEService *pService = pServer->createService(SERVICE_UUID);                                 // Create the BLE Service
 pTxCharacteristic                     =                                                      // Create a BLE Characteristic 
     pService->createCharacteristic(CHARACTERISTIC_UUID_TX, NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);                 
 NimBLECharacteristic * pRxCharacteristic = 
     pService->createCharacteristic(CHARACTERISTIC_UUID_RX, NIMBLE_PROPERTY::WRITE);
 pRxCharacteristic->setCallbacks(new MyCallbacks());
 pService->start(); 
 BLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
 pAdvertising->addServiceUUID(SERVICE_UUID); 
 pAdvertising->setName(BLEbroadcastName);    
 pServer->start();                                                                            // Start the server  Nodig??
 pServer->getAdvertising()->start();                                                          // Start advertising
 TekstSprint("BLE Waiting a client connection to notify ...\n"); 
}
//                                                                                            //

```

h2zero, 2024-12-23:
Glad you got it working. The required changes are quite minimal, also good to see you are using the `override` specifier as that will help catch many issues like this.

closed 2025-05-19 by h2zero