h2zero/NimBLE-Arduino #815: New NimBLEDevice::setPower API can receive dBm values that exceeds the official

Abasz, 2024-12-20:
From the official ESP32 docs I understand that:

_"The ESP32 Bluetooth LE has 8 transmit power levels, corresponding to -12 ~ 9 dBm of transmit power, with a 3 dBm interval. The controller software limits the transmit power and selects the power level according to the corresponding power level declared by the product."_ (https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/bt/ble.html)

Based on this I wonder whether it is now possible the exceed this amount with this library, e.g. by providing 30dBm to the `setPower` method. 

In your code there is a comment: `// round up to the next multiple of 3 to be able to target 20dbm`

But actually a few lines down you convert back to the `esp_power_level_t` where the max is 9dBm I think.

Thanks!

h2zero, 2024-12-20:
The controller will just set whatever it's own maximum is, calling getPower will reveal the actual power level set.

Abasz, 2024-12-20:
I see. Thanks!

closed 2024-12-20 by Abasz